mpromonet/webrtc-streamer #83: webrtc-streamer without -a dumps core

saket424, 2017-11-15:
root@ubulap2:/home/anand/w/webrtc-streamer# ./webrtc-streamer 
Logger level:4
[000:000] [7378] (audio_device_impl.cc:255): Failed to create the platform specific ADM implementation.
[000:001] [7378] (audio_device_impl.cc:255): Failed to create the platform specific ADM implementation.
[000:001] [7378] (voe_base_impl.cc:165): Init() failed to create the ADM

#
# Fatal error in ../../media/engine/webrtcvoiceengine.cc, line 287
# last system error: 0
# Check failed: 0 == voe_wrapper_->base()->Init(adm_.get(), apm(), decoder_factory_) (0 vs. -1)
# 
#

==== C stack trace ===============================

./webrtc-streamer() [0x4992d2]
./webrtc-streamer() [0x4690c3]
./webrtc-streamer() [0x4673da]
./webrtc-streamer() [0x7f76b1]
./webrtc-streamer() [0x4869a9]
./webrtc-streamer() [0x486df3]
./webrtc-streamer() [0x7f63a7]
./webrtc-streamer() [0x62e485]
./webrtc-streamer() [0x62fcfe]
./webrtc-streamer() [0x62e282]
./webrtc-streamer() [0x466278]
./webrtc-streamer() [0x4660f0]
./webrtc-streamer() [0x466667]
./webrtc-streamer() [0x40eab7]
./webrtc-streamer() [0x43ea57]
16: __libc_start_main
./webrtc-streamer() [0x40da69]
Aborted (core dumped)

Works fine when -a is specified

root@ubulap2:/home/anand/w/webrtc-streamer# ./webrtc-streamer -a
Logger level:4
[000:000] [7699] (voe_base_impl.cc:195): Init() failed to set the default output device
[000:001] [7699] (voe_base_impl.cc:198): Init() failed to initialize the speaker
[000:001] [7699] (voe_base_impl.cc:204): Init() failed to set the default input device
[000:001] [7699] (voe_base_impl.cc:207): Init() failed to initialize the microphone
[000:001] [7699] (voe_base_impl.cc:212): Init() failed to query stereo playout mode
[000:001] [7699] (voe_base_impl.cc:215): Init() failed to set mono/stereo playout mode
[000:001] [7699] (voe_base_impl.cc:227): Init() failed to set mono/stereo recording mode
[000:001] [7699] (audio_device_generic.cc:38): BuiltInAECIsAvailable: Not supported on this platform
[000:001] [7699] (audio_device_generic.cc:48): BuiltInAGCIsAvailable: Not supported on this platform
[000:001] [7699] (apm_helpers.cc:69): Failed to set AGC mode in ADM: 1
[000:001] [7699] (audio_device_generic.cc:58): BuiltInNSIsAvailable: Not supported on this platform
[000:001] [7699] (adm_helpers.cc:47): Unable to set recording device.
[000:001] [7699] (adm_helpers.cc:91): Unable to set playout device.
HTTP Listen at 0.0.0.0:8000

-a is not listed as an available option in the help menu. what does -a do ?
./webrtc-streamer -h
./webrtc-streamer [-H http port] [-S[embeded stun address]] [-t [username:password@]turn_address] -[v[v]]  [url1]...[urln]
./webrtc-streamer [-H http port] [-s[externel stun address]] [-t [username:password@]turn_address] -[v[v]] [url1]...[urln]
./webrtc-streamer -V
	 -H hostname:port   : HTTP server binding (default 0.0.0.0:8000)
	 -w webroot         : path to get files
	 -c sslkeycert      : path to private key and certificate for HTTPS
	 -S[stun_address]    : start embeded STUN server bind to address (default 0.0.0.0:3478)
	 -s[stun_address]   : use an external STUN server (default stun.l.google.com:19302)
	 -t[username:password@]turn_address : use an external TURN relay server (default disabled)
	 [url]              : url to register in the source list
	 -v[v[v]]           : verbosity
	 -V                 : print version

saket424, 2017-11-15:
case 'a': audioLayer = optarg ? (webrtc::AudioDeviceModule::AudioLayer)atoi(optarg) : webrtc::AudioDeviceModule::kDummyAudio; break;

Can you explain what is the correct usage of -a

Thanks

mpromonet, 2017-11-16:
Hi,
It disable audio capture.
Best regards
Michel

kibot914, 2017-11-17:
[image]

Also got the same issue with Docker, is there any help to resolve ? thanks

closed 2017-11-19 by mpromonet

mpromonet, 2017-11-19:
this is a duplicate of #78

flystar123, 2019-05-28:
#
# Fatal error in: ../../media/engine/adm_helpers.cc, line 38
# last system error: 88
# Check failed: 0 == adm->Init() (0 vs. -1)
# Failed to initialize the ADM.
~
~

I have encountered such a thorny problem, who can tell me how to solve it？Thanks

mpromonet, 2019-05-28:
Hi flystar123,
Did you tried to add '-a' option ?
Best Regards,
Michel.

fukemy, 2020-07-07:
i facing same problem here:
```
(voe_base_impl.cc:217): Init() failed to set mono/stereo playout mode
(voe_base_impl.cc:229): Init() failed to set mono/stereo recording mode
(audio_device_generic.cc:38): BuiltInAECIsAvailable: Not supported on this platform
(audio_device_generic.cc:48): BuiltInAGCIsAvailable: Not supported on this platform
```

mpromonet, 2020-07-07:
Hi fukemy,

The logs doesnot show "failed to create the ADM" and then it doesnot seems to crash with core dump.
What means same problem ?

Best Regards,
Michel.

lukelu0520, 2020-07-28:
Got the same issue on Linux-x86_64, we guess the reason is because there is no audio capture device on the machine.
by ./webrtc-streamer rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
Without -a flag, it prints:

Version:v0.3.0/Linux-x86_64 civetweb@v1.11 webrtc@6c5f787741-dirty live555helper@b4d86f2
{
	"urls" : 
	{
		"rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov" : 
		{
			"video" : "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"
		}
	}
}Logger level:3

#
# Fatal error in: ../../rtc_base/synchronization/mutex.h, line 72
# last system error: 88
# Check failed: holder != current (140171966275328 vs. 140171966275328)
# Aborted (core dumped)

====

Add -a flag make it working, will add an audio capture device to see if it works on Linux-x86_64.

-- Luke

weijunying2019102969, 2023-11-22:
> ```
> #
> # Fatal error in: ../../media/engine/adm_helpers.cc, line 38
> # last system error: 88
> # Check failed: 0 == adm->Init() (0 vs. -1)
> # Failed to initialize the ADM.
> ```
> 
> I have encountered such a thorny problem, who can tell me how to solve it？Thanks

Did you solve it? It happened to me ,too

weijunying2019102969, 2023-11-22:
> Hi flystar123, Did you tried to add '-a' option ? Best Regards, Michel.

what is "-a" mean? which should I add?

mpromonet, 2023-11-22:
Hi @weijunying2019102969,

This issue was 3 years old.... code and webrtc have changed... what is the release you are using ?
-a disable audio layer.

Best Regards,
Michel.

commit ffc0c085 (2017-11-19): fix #83: fix audio capture initialization using ALSA instead of Pulse