mpromonet/webrtc-streamer #84: AddSink Segmentation fault (core dumped)

abhisit, 2017-11-16:
**Question**
How to I fix this issue please?
_AddSink
Segmentation fault (core dumped)_

**Environment / typing / happen order**
All of terminals are running on the same machine Ubuntu 16.04.3 x86_64.
(1) -> (2) -> (3) -> (4) -> (5)

**Terminal 1** 
Pre-compiled webrtc-streamer v0.0.6 x86_64 official release.

(1) $ ./webrtc-streamer rtsp://USER:PASSWORD@IP:.IP.IP.IP:PORT/Streaming/Channels/1/
(5) _=> AddSink
Segmentation fault (core dumped)_

**Terminal 2** 
Nodejs v8.9.1

(2) # node [feeder.js](https://gist.github.com/abhisit/5c633530c8921f9644a151f14155f904)
(3) Ctrl+C
(4) # node feeder.js
HTTP call http://IP.IP.IP.IP:PORT/createOffer?peerid=0.8668596165446638&url=rtsp%3A%2F%2Fndrs%3Andrstest1%40IP.IP.IP.IP%3APORT%2FStreaming%2FChannels%2F1%2F
/root/webrtc-testing/feeder.js:22
			console.log("HTTP code:"+ response.statusCode);
			                                   ^

TypeError: Cannot read property 'statusCode' of undefined

mpromonet, 2017-11-16:
Hi,
Maybe gdb could give more information
Best regards
Michel

k13elle, 2017-11-17:
Hi Michel,

I also face this problem
I already make a little bit change this [peerid random](https://github.com/mpromonet/webrtc-streamer/blob/86f941e4f6dbc6d2d12fa5ccd8cd104feea1e64c/html/janusvideoroom.js#L96) by multiply with `Date.now()`, but it won't work.

Best regards,
Nathapon.

mpromonet, 2017-11-19:
@abhisit 
I am not able to reproduce your problem, you should look to the coredump.
Also the log "AddSink" doesnot exists anymore, did you tried updating the code.
Best Regards,
Michel.

abhisit, 2017-11-20:
@mpromonet Thank you for your help.

No, I did not try up to date code, I will try to use up to date code.

I thought, the official release is compiled on Debian and I am running on Ubuntu 16.04.2, does it problem?

Anyway, [here is the webrtc-streamer v0.0.7 gdb log file](https://gist.github.com/abhisit/f27c67e029a1e44ba32dc5050ac85338).
Short log:
[New Thread 0xb49df450 (LWP 2205)]
=> AddSink
[279:028] [1623] (voe_base_impl.cc:453): StopSend() failed to stop recording
[279:028] [1623] (audio_send_stream.cc:258): AudioSendStream::Stop failed with error: -1
[279:033] [1623] (voe_base_impl.cc:453): StopSend() failed to stop recording
[279:033] [1623] (audio_send_stream.cc:258): AudioSendStream::Stop failed with error: -1
[279:037] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:037] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:038] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:038] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:039] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:039] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:040] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[279:041] [1627] (basicpacketsocketfactory.cc:59): UDP bind failed with error 99
[New Thread 0xb41c1450 (LWP 2206)]
[New Thread 0xb39c1450 (LWP 2207)]

Thread 8 "civetweb-worker" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb63d3450 (LWP 1632)]
0x0008172c in rtc::Thread::ProcessMessages(int) ()
(gdb)

mpromonet, 2017-11-21:
@abhisit 
So maybe the problem is fixed by the commit 715e5e9eeddec4101a1e8d19dcb099897e60fad8.
However I have some difficulties to build on Ubuntu 14.04 (it's OK with 16.04) since WebRTC update its sysroot from jessie to stretch, I will push a new release when this will be fixed.
Best Regards,
Michel.

closed 2017-11-26 by mpromonet