mpromonet/webrtc-streamer #86: question: can you explain what sendstream.html is intended for ?

saket424, 2017-11-18:
Hi Michel,
I just noticed a recent commit from you 60c9922aaacccd63eec6a8bbb15a806946bab99e

Is this new functionality and if yes can you explain what the new functionality is supposed to do, relay a stream somewehere? 
Usage says
?video=<WebRTC video stream name>&audio=<WebRTC audio stream name>&options=<WebRTC options>

It would help me if you could provide a concrete example that explains how to use this 

Thanks

mpromonet, 2017-11-18:
Hi sacket,
This is in progress, you should give à webrtc stream to receive, maybe this Will be optionnal.
Next it should send the local stream to the server, that will write to à v4l2 output device.
This is the idea, but it needs work
Best regards
Michel

saket424, 2017-11-18:
That is what I thought the intent was. Cool, will look for progress on this topic in future commits
Thanks

closed 2020-10-09 by mpromonet