mpromonet/webrtc-streamer #87: Stream /dev/v4l/by-id/usb-INOGENI_2633-INOGENI_4K2USB3_01C92633-video-index0 device

gaspa85, 2017-12-01:
Hi,

is possibile to stream /dev/v4l/by-id/usb-INOGENI_2633-INOGENI_4K2USB3_01C92633-video-index0?

i have try to start server with this command

./webrtc-streamer -vvv /dev/v4l/by-id/usb-INOGENI_2633-INOGENI_4K2USB3_01C92633-video-index0

and after open browser with this url

http://10.0.2.86:8000/webrtcstream.html?/dev/v4l/by-id/usb-INOGENI_2633-INOGENI_4K2USB3_01C92633-video-index0

the server return this log
[log.txt](https://github.com/mpromonet/webrtc-streamer/files/1521036/log.txt)

thanks for your times

Marco Gasparini

mpromonet, 2017-12-01:
Hi Marko,
You don't need to give the url to capture, webrtc-streamer enumerate using WebRTC SDK available capture device.
You can get the list using http://10.0.2.86:8000/getVideoDeviceList.
Next should should be able to display http://10.0.2.86:8000/webrtcstream.html?<with the name of your device> or http://10.0.2.86:8000/webrtcstream.html?video=<with the name of your device>.
Best Regards,
Michel.

gaspa85, 2017-12-04:
thanks so mutch Michel, find periferal and start stream.
Marco G.

closed 2017-12-04 by gaspa85