mpromonet/webrtc-streamer #89: avcodec_send_packet error: -1094995529

Bhlowe, 2017-12-01:
Getting the following errors when running with my IP camera:

[h264 @ 0x7f2dec019ee0] no frame!
[024:323] [67] (h264_decoder_impl.cc:327): avcodec_send_packet error: -1094995529
NOTIFY failed
repeats often

Video looks great.. ! but the errors clog the error log and may be something going on. 

Sending you the docker command with rtsp URL/password to you via PM.

Bhlowe, 2017-12-08:
I ended up discarding NAL units of type NAL_UNIT_TYPE_AUD and NAL_UNIT_TYPE_SEI in rtspcapturer.cpp and the errors/warnings went away.  I can submit a pull request, but not sure if the "fix" is the best solution... but didn't want to debug the webrtc code.

closed 2017-12-08 by Bhlowe

mpromonet, 2017-12-10:
Hi,
Probably this is more ffmpeg code, or ffmpeg expected theses frames should be aggregate with others. This is the case for SPS/PPS that are refused by ffmpeg when send without a image frame.
You could disable the ffmpeg warning adding somewhere in the code
           `av_log_set_level(AV_LOG_QUIET);`
Best Regards,
Michel.