mpromonet/webrtc-streamer #90: Change Streaming Video Resolution FPS

gaspa85, 2017-12-04:
hi Michel,

is possibile to change Streaming resolution and fps of /dev/v4l/...

thanks 

Marco

mpromonet, 2018-01-14:
Hi Marco,
The logic is in WebRTC SDK itself, maybe using some media constraints option you could make WebRTC to makes a different choice.
I wrapped bandwidth option, perhaps you can use it to reduce fps ?
Best Regards,
Michel.

gaspa85, 2018-01-15:
Hi Michel,
depend, in this application is necessary 30 fps.
But if you wrapped all Camera capability i select more correct for my purpose.

thanks Michel

LanderN, 2018-09-10:
Hello!
Any updates on this? I'm using this on a Raspberry Pi and I seem unable to get a higher resolution than 640x480.
How would I add constraints like you suggested?

LanderN, 2018-09-13:
I figured it out! Leaving this here in case it helps someone else:
In `PeerConnectionManager::CreateVideoTrack`:
```cpp
#include "api/test/fakeconstraints.h"

[...]

webrtc::FakeConstraints constraints;
constraints.AddMandatory(webrtc::MediaConstraintsInterface::kMinWidth, "1920");
constraints.AddMandatory(webrtc::MediaConstraintsInterface::kMinHeight, "1080");
rtc::scoped_refptr<webrtc::VideoTrackSourceInterface> videoSource =
    peer_connection_factory_->CreateVideoSource(std::move(capturer), &constraints);

[...]
```

mpromonet, 2018-09-13:
Hi LanderN,

I tried this with kMaxWidth/kMaxHeight to limit the resolution size, but it seems that WebRTC lib doesnot takes into account.
Maybe this was recently added.

Is it working in your situation ?

Best Regards,
Michel.

closed 2018-09-27 by mpromonet