mpromonet/webrtc-streamer #92: Errors related to "undefined reference to `std::__cxx11" and "undefined reference to `X for std::__cxx11"

abhikandoi, 2018-01-03:
Here is a part of the console log while the last command to build `webrtc-streamer` binary runs:  

```libWebRTC__Release.a(echo_remover_metrics.o):echo_remover_metrics.cc:(.text._ZN6webrtc18EchoRemoverMetrics6UpdateERKNS_8AecStateERKSt5arrayIfLm65EES7_+0x17a): more undefined references to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)' follow
libWebRTC__Release.a(neteq.o): In function `webrtc::NetEq::Config::ToString[abi:cxx11]() const':
neteq.cc:(.text._ZNK6webrtc5NetEq6Config8ToStringB5cxx11Ev+0x65): undefined reference to `VTT for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >'
neteq.cc:(.text._ZNK6webrtc5NetEq6Config8ToStringB5cxx11Ev+0x10e): undefined reference to `vtable for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >'
neteq.cc:(.text._ZNK6webrtc5NetEq6Config8ToStringB5cxx11Ev+0x190): undefined reference to `vtable for std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >'
neteq.cc:(.text._ZNK6webrtc5NetEq6Config8ToStringB5cxx11Ev+0x31a): undefined reference to `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::str() const'
neteq.cc:(.text._ZNK6webrtc5NetEq6Config8ToStringB5cxx11Ev+0x32f): undefined reference to `VTT for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >'```

I am using the following two commands to build webrtc:
1. `cd ../webrtc/src`
2. ```gn gen out/Release --args='is_debug=false rtc_use_h264=true ffmpeg_branding="Chrome" rtc_include_tests=false rtc_enable_protobuf=false use_custom_libcxx=false use_ozone=true rtc_include_pulse_audio=false rtc_build_examples=false'```  
3. `ninja -C out/Release jsoncpp rtc_json webrtc`

And this is my Makefile for the modified webrtc-streamer that I am using:
https://github.com/abhikandoi/webrtc-streamer/blob/master/Makefile

I know this is some sort of linker issue where std and c++11 are being used, but I am unable to figure out how to fix this.
This happens when I run it inside the docker image `nvidia/cuda:8.0-cudnn5-devel-ubuntu14.04` but doesn't happen on my system which is a ubuntu 16.04.

abhikandoi, 2018-01-03:
@mpromonet please see if you can guide me. I can provide any information that is required. Can you please guide me here?

mpromonet, 2018-01-03:
HI abhikandoi,
When WebRTC moved the sysroot from jessie  to stetch, I was not able to build with 14.04 and move to 16.04.
There is probably several layer of problems between ABI , mixing clang/g++ (that needs use_custom_libcxx=false).
I think it should be easier to use `nvidia/cuda:8.0-cudnn5-devel-ubuntu16.04`
Best Regards,
Michel.

mpromonet, 2018-05-23:
I close the issue because now build for x86_64 doesnot use anymore sysroot provided by WebRTC

closed 2018-05-23 by mpromonet