mpromonet/webrtc-streamer #95: No image on localhost

medzik1993, 2018-02-01:
Hi, a have problem with rtsp image. 
Is should by simple. Install docker on my windows, paste docker command 
docker run -p 8000:8000 -it mpromonet/webrtc-streamer -n raspicam -u <My rtsp stream> 
and go localhost:8000. In console I see that browser connect to docker, by in browser there is no video. docker receive only few packages and nothing happend, 
Of course I tried many rtsp streams, and public and local, but stil there is no image in browser. 

What i doing wrong ? 
Thanks for help.

mpromonet, 2018-02-01:
Hi medzik1993,
Did you see in the docker console RTSP negociation ? some DESCRIBE, PLAY,... ?
Regards,
Michel.

tarag, 2018-03-15:
Hello,

I have the same issue. I am running latest docker image (v0.0.9-5-ga717abe) on macOS. While the demo page works great (https://webrtc-streamer.herokuapp.com), using either a LAN RSTP camera or one working 

on the demo page does not work.

You can see attached logs (-vvv), there seems to be some connection but afterwards issue with the data received. 

[errlog.txt](https://github.com/mpromonet/webrtc-streamer/files/1814584/errlog.txt)
[errlog2.txt](https://github.com/mpromonet/webrtc-streamer/files/1814585/errlog2.txt)

I tried docker release 0.0.8 but nothing happened at all...

Thanks for the help.

mpromonet, 2018-03-15:
Hi Gautier,
From your log, we can see that you never receive IDR frame, probably big frames are lost by your network.
You can try to connect using option `rtptransport=tcp`
Best Regards,
Michel.

tarag, 2018-03-16:
Thank you Michel for your answer.

How do I pass this option to the program? I could not understand from the source.

Right now I am launching with this command:
`docker run -p 8000:8000 -it mpromonet/webrtc-streamer -vvv -n Cam  -u "rtsp://85.255.175.244/h264"`

Yet I am a bit surprised by the reason you give to the issue because one of the streams I am trying it on is a LAN camera with a gigabit switch in between the cam and the computer and almost no other traffic...

mpromonet, 2018-03-16:
Hi,
I read too quickly, there is some I-frame : 
```
[009:279] [64] (rtspvideocapturer.cpp:107): RTSPVideoCapturer:onData size:113031 ts:1521103369776
[009:280] [64] (rtspvideocapturer.cpp:150): RTSPVideoCapturer:onData IDR
```
The options could be set in webrtcstreamer.json or in the html/js that call WebRtcStreamer.connect(videourl, audiourl, options)
Best Regards,
Michel.

chenyunchen, 2018-04-02:
Same question here.
Get the error on my browser(OSX).
`oniceconnectionstatechange  state: failed`
But all works fine on my ubuntu server

closed 2018-05-23 by mpromonet