mpromonet/webrtc-streamer #96: how to get input stream from /dev/video1 loopback device using webrtc-streamer commandline

danishnazir, 2018-02-05:
Hi First Thanks for this amazing repository 
I have a question which i would like to be answered
-> i have a loopback device at /dev/video1 now what i want to do is i need to post the video stream so that i can access it outside from my local network can you help me here that how i would be able to achieve it? do i need to first create the rtsp url and the give it to webrtc-streamer with proper stun/turn servers? if yes then how will i be able to access them outside the network?

mpromonet, 2018-02-05:
Hi Danish,
The V4L2 device are available using there device name, you should see it using `/getVideoDeviceList` API.
Next you should be able to reach it using `/webrtcstream.html?video=<devicename>`.
By default ICE url is the default google one (stun.l.google.com:19302), but you can chosse it using `-S` option.
Best Regards,
Michel.

danishnazir, 2018-02-20:
Hi thanks for replying.

I have installed and deployed the server locally when ever i open the page i.e http://10.0.11.174:8080/webrtcstream.html?rtsp://217.17.220.110/axis-media/media.amp
8080/getIceServers Failed to load resource: the server responded with a  status of 404 (Not Found) 
i have started webrtc streamer like this
./webrtc-streamer rtsp://217.17.220.110/axis-media/media.amp
and it is returning me this
Logger level:4
[000:000] [7202] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:000] [7202] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:000] [7202] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000

Please guide me here how to solve this Thanks again sir @mpromonet

mpromonet, 2018-02-25:
Hi Danish,
As you mention 2 ports 8000 & 8080, I guess you are serving html page using an external HTTP server.
In such a case you should path the url of the webrtc-streamer (port 8000 in your case) to the WebRtcStreamer object.
To simplify this, I add in the commit edbb470f503f18f214d5cabbf9490c0f6c51f646 a configuration file webrtcstremer.json that could be customize to set this.
Default is 
```
var webrtcConfig = {
	url: "",
	options: "rtptransport=tcp&timeout=60"
}
```
webrtcConfig.url is empty when page served by webrtc-streamer internal http server, and you could set it give the webrtc-streamer url when page are served by an external HTTP server.
Regards,
Michel.

danishnazir, 2018-03-02:
@mpromonet yup thanks now i am able to connect with the server! other problem is that i am unable to get the stream of my ip camera displayed on the browser here are the logs from webrtcstream.html page
`{video: "rtsp://admin:newtech148@192.168.0.198"}
webrtcstreamer.js:45 Get IceServers
sendrequest.js:3 HTTP call http://10.0.11.174:8000/getIceServers
webrtcstreamer.js:122 createPeerConnection  config: {"iceServers":[{"url":"stun:stun.l.google.com:19302"}]} option:{"optional":[{"DtlsSrtpKeyAgreement":true}]}
webrtcstreamer.js:168 Created RTCPeerConnnection with config: {"iceServers":[{"url":"stun:stun.l.google.com:19302"}]}option:{"optional":[{"DtlsSrtpKeyAgreement":true}]}
webrtcstreamer.js:102 Create offer:{"type":"offer","sdp":"v=0\r\no=- 1692714314855188127 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video data\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:T6CP\r\na=ice-pwd:7YpZaIwrVwo+DkKwXR2q8Zrg\r\na=ice-options:trickle\r\na=fingerprint:sha-256 0A:26:ED:E0:29:93:70:7B:6A:B8:B0:2C:A0:21:7E:68:9E:9A:99:FC:A2:E6:61:DF:BA:10:43:10:F6:AA:AC:D3\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 125 127\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:T6CP\r\na=ice-pwd:7YpZaIwrVwo+DkKwXR2q8Zrg\r\na=ice-options:trickle\r\na=fingerprint:sha-256 0A:26:ED:E0:29:93:70:7B:6A:B8:B0:2C:A0:21:7E:68:9E:9A:99:FC:A2:E6:61:DF:BA:10:43:10:F6:AA:AC:D3\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:102 red/90000\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=102\r\na=rtpmap:127 ulpfec/90000\r\nm=application 9 DTLS/SCTP 5000\r\nc=IN IP4 0.0.0.0\r\na=ice-ufrag:T6CP\r\na=ice-pwd:7YpZaIwrVwo+DkKwXR2q8Zrg\r\na=ice-options:trickle\r\na=fingerprint:sha-256 0A:26:ED:E0:29:93:70:7B:6A:B8:B0:2C:A0:21:7E:68:9E:9A:99:FC:A2:E6:61:DF:BA:10:43:10:F6:AA:AC:D3\r\na=setup:actpass\r\na=mid:data\r\na=sctpmap:5000 webrtc-datachannel 1024\r\n"}
sendrequest.js:3 HTTP call http://10.0.11.174:8000/call?peerid=0.5545340692036795&url=rtsp%3A%2F%2Fadmin%3Anewtech148%40192.168.0.198
webrtcstreamer.js:211 offer: {"sdp":"v=0\r\no=- 2271082767691875708 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE data\r\na=msid-semantic: WMS\r\nm=audio 0 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:kN0Z\r\na=ice-pwd:AqL+NEoSL82yC7PS4CLTj1BD\r\na=ice-options:trickle\r\na=fingerprint:sha-256 FF:09:00:95:D6:D7:91:1A:AC:11:13:D8:E1:CC:C5:45:40:48:D9:07:43:6A:3F:0E:A1:5B:B0:BF:B0:53:E3:23\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=inactive\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 0 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 125 127\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:pR7K\r\na=ice-pwd:C5DkPIVjvO60oopuY+wxVKRz\r\na=ice-options:trickle\r\na=fingerprint:sha-256 FF:09:00:95:D6:D7:91:1A:AC:11:13:D8:E1:CC:C5:45:40:48:D9:07:43:6A:3F:0E:A1:5B:B0:BF:B0:53:E3:23\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-tra [... cut ...]
webrtcstreamer.js:133 oniceconnectionstatechange  state: connected
webrtcstreamer.js:161 local datachannel open
webrtcstreamer.js:148 remote datachannel created:{"isTrusted":true}
webrtcstreamer.js:151 remote datachannel open
webrtcstreamer.js:155 remote datachannel recv:"ServerDataChannel open"
webrtcstreamer.js:165 local datachannel recv:"ClientDataChannel open"
webrtcstreamer.js:133 oniceconnectionstatechange  state: completed`
 can you please guide me here that which thing i am missing here or whats the problem thanks for your time i appreciate it.