mpromonet/webrtc-streamer #97: Why I cannot see anything on https://webrtc-streamer.herokuapp.com/ ??

secombrian, 2018-02-07:
As title, 
Here is the console log from my chrome as follow:

sendrequest.js:3 HTTP call /getMediaList
sendrequest.js:3 HTTP call /version
webrtcstreamer.js:44 Get IceServers
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/getIceServers
webrtcstreamer.js:118 createPeerConnection  config: {"iceServers":[{"url":"stun:stun.l.google.com:19302"}]} option:{"optional":[{"DtlsSrtpKeyAgreement":true}]}
webrtcstreamer.js:164 Created RTCPeerConnnection with config: {"iceServers":[{"url":"stun:stun.l.google.com:19302"}]}option:{"optional":[{"DtlsSrtpKeyAgreement":true}]}
webrtcstreamer.js:98 Create offer:{"type":"offer","sdp":"v=0\r\no=- 7160422080670743163 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video data\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:5J7o\r\na=ice-pwd:xvUqgmi+F4h90All7t14tcXP\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F2:A9:40:C9:8E:94:1F:75:0E:0C:D1:2C:EB:F6:3F:62:00:CA:3A:D3:7C:C8:34:7A:6F:87:F9:8B:EA:98:70:35\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 124 127 123 125 107 108\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:5J7o\r\na=ice-pwd:xvUqgmi+F4h90All7t14tcXP\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F2:A9:40:C9:8E:94:1F:75:0E:0C:D1:2C:EB:F6:3F:62:00:CA:3A:D3:7C:C8:34:7A:6F:87:F9:8B:EA:98:70:35\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=64001f\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:124 rtx/90000\r\na=fmtp:124 apt=102\r\na=rtpmap:127 H264/90000\r\na=rtcp-fb:127 ccm fir\r\na=rtcp-fb:127 nack\r\na=rtcp-fb:127 nack pli\r\na=rtcp-fb:127 goog-remb\r\na=rtcp-fb:127 transport-cc\r\na=fmtp:127 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f\r\na=rtpmap:123 rtx/90000\r\na=fmtp:123 apt=127\r\na=rtpmap:125 red/90000\r\na=rtpmap:107 rtx/90000\r\na=fmtp:107 apt=125\r\na=rtpmap:108 ulpfec/90000\r\nm=application 9 DTLS/SCTP 5000\r\nc=IN IP4 0.0.0.0\r\na=ice-ufrag:5J7o\r\na=ice-pwd:xvUqgmi+F4h90All7t14tcXP\r\na=ice-options:trickle\r\na=fingerprint:sha-256 F2:A9:40:C9:8E:94:1F:75:0E:0C:D1:2C:EB:F6:3F:62:00:CA:3A:D3:7C:C8:34:7A:6F:87:F9:8B:EA:98:70:35\r\na=setup:actpass\r\na=mid:data\r\na=sctpmap:5000 webrtc-datachannel 1024\r\n"}
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/call?peerid=0.5720020784158875&url=Norway
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
sendrequest.js:3 HTTP call https://webrtc-streamer.herokuapp.com:/addIceCandidate?peerid=0.5720020784158875
webrtcstreamer.js:177 End of candidates.
request.min.js:2 POST https://webrtc-streamer.herokuapp.com/addIceCandidate?peerid=0.5720020784158875 404 (Not Found)
(anonymous) @ request.min.js:2
tryCallTwo @ request.min.js:10
doResolve @ request.min.js:10
Promise @ request.min.js:10
doRequest @ request.min.js:2
sendRequest @ sendrequest.js:9
WebRtcStreamer.onIceCandidate @ webrtcstreamer.js:174
pc.onicecandidate @ webrtcstreamer.js:121
webrtcstreamer.js:203 offer: {"sdp":"v=0\r\no=- 2669590926496689805 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video data\r\na=msid-semantic: WMS rtsp://217.17.220.110/axis-media/media.amp\r\nm=audio 57095 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 172.17.133.74\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=candidate:1300230808 1 udp 2122260223 172.17.133.74 57095 typ host generation 0 network-id 1 network-cost 50\r\na=ice-ufrag:WoP5\r\na=ice-pwd:fMcoUg63xKPs1XsmVXfwMc+o\r\na=ice-options:trickle\r\na=fingerprint:sha-256 48:F4:F0:18:E7:C4:F3:94:83:C2:8A:56:96:15:FE:79:EE:45:41:EB:A2:AF:A6:80:99:94:58:27:2E:D7:6A:EB\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=sendonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\na=ssrc:676260080 cname:LOLkE4Is7chIFeo6\r\na=ssrc:676260080 msid:rtsp://217.17.220.110/axis-media/media.amp audio_label\r\na=ssrc:676260080 mslabel:rtsp://217.17.220.110/axis-media/media.amp\r\na=ssrc:676260080 label:audio_label\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 102 124 127 123 125 107 108\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:WoP5\r\na=ice-pwd:fMcoUg63xKPs1XsmVXfwMc+o\r\na=ice-options:trickle\r\na=fingerprint:sha-256 48:F4:F0:18:E7:C4:F3:94:83:C2:8A:56:96:15:FE:79:EE:45:41:EB:A2:AF:A6:80:99:94:58:27:2E:D7:6A:EB\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:8 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=sendonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:124 rtx/90000\r\na=fmtp:124 apt=102\r\na=rtpmap:127 H264/90000\r\na=rtcp-fb:127 goog-rem [... cut ...] ","sdpMLineIndex":0}
webrtcstreamer.js:221 Adding ICE candidate :{"candidate":"candidate:66850408 1 tcp 1518280447 172.17.133.74 55355 typ host tcptype passive generation 0 ufrag Km/V network-id 1 network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
webrtcstreamer.js:223 addIceCandidate OK
webrtcstreamer.js:223 addIceCandidate OK
webrtcstreamer.js:223 addIceCandidate OK
webrtcstreamer.js:129 oniceconnectionstatechange  state: failed
webrtcstreamer.js:129 oniceconnectionstatechange  state: failed

danishnazir, 2018-02-21:
same issue here did you resolve it?

secombrian, 2018-02-22:
Hi Michel, 
The link is be redirected to http://pc1.herokuappp.com/

secombrian, 2018-02-22:
@danishnazir  not yet, but I can see the rtsp steam at Chrome on my android phone via LTE. Maybe it's about firewall and NAT difference.

danishnazir, 2018-02-22:
@mpromonet the problem  is i push the stream of the webrtc-stream server using ./webrtc-... rtsp_url which runs fine on 0.0.0.:8080 then i open the webpage and i see nothing! i got curious so i inspected the video element and i see that the source is set to null! any help would be appreciated thanks