mpromonet/webrtc-streamer #99: Is Decoding and reencoding needed?

Bhlowe, 2018-02-15:
If I have an H.264 stream from an IP camera and the web client supports H.264 is there a way to send the packets directly to the client browser without decoding and encoding? 

Or does the webrtc need to be able to change the quality on the fly so each outgoing stream needs their own encoded version? 

I'm seeing some video corruption that I don't see when connected to VLC when my IP camera is on a wireless/UDP connection. Just trying to figure out the best way to solve. Would like to just decode the rtsp packets and send to the client directly. Possible?

Thanks

mpromonet, 2018-02-17:
Hi Brad,

You can try to remove the decoding code sending directly the H264 frame to webrtc::VideoCapturer, maybe it support other format than I420 format. If not you will need to overide it or using lower level class.

An other approach could be to use janus-gateway that doesnot transcode the RTP packet and relay them to webrtc.

If you have some RTP packet loss, you can try to enable `rtptransport=tcp`

Best Regards,
Michel.

rgaufman, 2018-05-06:
I'm having this problem too and 1 IP camera is taking 100% CPU.

Can you please add a command line option to force decoding off? - Alternatively, can you please share details where to disable this in the source code?

I will check out janus-gateway, thank you.

mpromonet, 2018-05-09:
Hi Roman,
Using 100% of CPU (1?) could depends on the hardware you are using and also the options used to compiled WebRTC.
My understanding is that WebRTC library used I420 as a pivot format, capturer should provide to I420, and output should consume I420. But maybe there is other way to avoid this, you can contact WebRTC team.
You should also note that this conversion give a better decoupling between provider and consumer, it give more interoperability (without the web browser need to support the camera compression, maybe now H264 high profile is supported?) and bandwidth adaptation, the input stream and output stream could have different compression format, geometry, frame per seconds,...
Best Regards,
Michel.

mpromonet, 2018-05-09:
Hi Roman,

Are you using a camera locally that is captured by WebRTC librairy ? In such a case the code that choose the format is the WebRTC library and it choose I420, you can set the format you like, it will change it.
Next the web browser negociate to choose a format (compressed like H264, VP8, VP9) that is used to send the stream.

In this situation there is no decoding & re-encoding, the camera provide a uncompressed format that is encoded for streaming there is only 1 encoding done by WebRTC librairy (and so the compilation parameter could have an impact if you are using hardware encoding or not).

If you like WebRTC capture H264 format from the camera you can try https://bugs.chromium.org/p/webrtc/issues/list, still now this seems not implemented.

Best Regards,
Michel.

zoolyka, 2019-08-23:
It would be really great to have a stream "passthrough" option to avoid transcoding like how its done in Janus. Even that would be acceptible to send nothing to the client if it does not announce H264, since there are only a few exceptions. I would prefer this project over Janus, since this provides only what I need in a very clean way, but the additional latency and CPU overhead of the transcodings makes it very limited for the purpuse what it meant for.

dottgonzo, 2019-09-01:
will be great if we could have the ability to not charge the cpu with the additional overhead of the transcoding. Without the option of h264->webrtc passthrought stream the software is limited in number of simultaneous peers and/or streams. I know that will be impossible to support any stream to any client, (because the best way to maximalize the codecs coverage is is to use h264 with main profile on as input format, and to support only chrome and safari). This (beautiful) project is unusable as streaming platform and is relegate only to stream private in-home camera if no massive trasmission of streams is allowed. To stream to let' say 1000 clients in hight resolution it needs many nodes with gpu and cpu support, instead of 1 simple node (if bandwidth allows it) without any sort of transcoding

openaudible, 2019-09-02:
Would it make sense to intercept the rtsp data and send it to a multicast socket... then if we are doing passthrough, skip the decode/encode and just send the h.264 data directly? Perhaps in webrtc::VideoCapturer?

This might also make it easier to have multiple servers sending out the same stream to larger audiences. 

I'm willing to try this. But not sure exactly where and how to make the changes.

mpromonet, 2019-09-05:
Hi openaudible,

I did not dive enouph in WebRTC SKD, my actual understanding is that the api to provide a video is 
`rtc::scoped_refptr<VideoTrackInterface> PeerConnectionFactory::CreateVideoTrack(const std::string& id, VideoTrackSourceInterface* source) `
and VideoTrackSourceInterface inherit from `rtc::VideoSourceInterface<VideoFrame>`
and VideoFrame cannot handle an encoded frame.

So I don't see how to give an encoded frame without modification in WebRTC or understand better the way it works.

Best Regards,
Michel.