mpromonet/webrtc-streamer #102: Compilation instructions unclear

REPRESSION, 2018-03-08:
Hello, could you please provide more information on how to compile your project ?
I get this for the first steps:
command not found: fetch
command not found: gn

cant install those programs with apt-get install : not found.

Thanks

mpromonet, 2018-03-08:
Hi,
This tools are from google, you can get it using: 

     git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git

You can find more information about how to use it in http://commondatastorage.googleapis.com/chrome-infra-docs/flat/depot_tools/docs/html/depot_tools_tutorial.html#_setting_up.

In order to build the project, you can look to the build configuration in https://github.com/mpromonet/webrtc-streamer/blob/master/.circleci/config.yml or https://github.com/mpromonet/webrtc-streamer/blob/master/.travis.yml.

Best Regards,

Michel.

REPRESSION, 2018-03-09:
Hi, thank you for your help, I progressed but I am now stuck at the build of the webrtc-streamer.
When doing `make WEBRTCROOT=./webrtc all tgz` I get a lot of errors of this type :
~/webrtc-streamer/src/someFile.cpp:XX: undefined reference to `rtc::SomeClass::SomeMethod()`
or ~/webrtc-streamer/src/someFile.cpp:XX: undefined reference to JSON::SomeClass:SomeMethod()`

seems like JSON and rtc are not found, I have no idea where my error is.

Thanks

REPRESSION, 2018-03-09:
Ok got it working, for anyone facing the same issue this is what fixed it for me :
do not use ninja from depot_tools but ninja-linux64

closed 2018-03-09 by REPRESSION