mpromonet/webrtc-streamer #104: Build new WebRTC , unclear about live555

Kaminight, 2018-03-11:
Hello everyone! I just started with WebRTC and I don't understand  this step:

**Build live555 to enable RTSP support(optional)
make SYSROOT=<path to WebRTC>/src/build/linux/debian_stretch_amd64-sysroot live555**

When Im trying make this step , makes me:
make: *** No rule to make target 'live555'.  Stop.

So, I should be download the LIVE555 for make this build or I should be find a file ccp inside from WebRTC.

Sorry for my ignorence, have a nice day.

mpromonet, 2018-03-12:
Hi Kaminight,

The target live555 is in the root Makefile of this project https://github.com/mpromonet/webrtc-streamer/blob/master/Makefile, it download live555 and install it in SYSROOT.
If the target is not found, I guess you are not in the root folder of this projet, but somewhere else ?
To understand the compilation step you can look at https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile

With current WebRTC release it is now debian_stretch_amd64-sysroot.

Best Regards,
Michel.

Kaminight, 2018-03-12:
¡Hola de nuevo! ¡Gracias por la ayuda! Me pondre a trabajar con lo que me mencionaste,  vere como trabaja...

Kaminight, 2018-03-13:
After trying execute for first time, makes me this error:
** Execute -->  ./webrtc-streamer -H 8000 -s stun.l.google.com:19302**
./webrtc-streamer: relocation error: ./webrtc-streamer: symbol _ZTVNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference

How I can fix , this problem... I just running in Ubuntu Server 14.04, or need update my server, because  I can't. For reasons over other functions.

mpromonet, 2018-03-13:
Hi  Kaminight ,
I moved from trusty to xenial because of this difference between WebRTC embeded sysroot and gcc.
You can try: 

- build WebRTC using  use_sysroot=false in gn arguments (this will probably have some impact)
- build using docker, there is a DockerFile in the project
- upgrade to xenial

Best Regards,
Michel.

mpromonet, 2018-04-23:
Hi Kaminight ,
Due to complexity to maintain compliance between WebRTC embededed sysroot and the build environnement I finally change the way to build WebRTC setting "use_sysroot=false".
Otherwise it is needed to move to artful.
Then I guess you should no more have the problem between libstdc++ for build & runtime since https://github.com/mpromonet/webrtc-streamer/commit/e1676266fbd034a150f39f7737171f1476d54443.

Because of this modification I close the issue. If you still have a problem, please open a new one.

Best Regards,
Michel.

closed 2018-04-23 by mpromonet