mpromonet/webrtc-streamer #105: Building webrtc under raspi has issues...

JudaismBot, 2018-03-16:
libc++abi seems to no longer be compatible...

https://reviews.llvm.org/D42242  tried a level of patches, but then seems to have an include issue with unwind.h

_URC_FATAL_PHASE1_ERROR and _URC_FATAL_PHASE2_ERROR should be enum values of _Unwind_Reason_Code, but this is defined in unwind-arm-common.h as:

typedef enum
  {
    _URC_OK = 0,       /* operation completed successfully */
    _URC_FOREIGN_EXCEPTION_CAUGHT = 1,
    _URC_END_OF_STACK = 5,
    _URC_HANDLER_FOUND = 6,
    _URC_INSTALL_CONTEXT = 7,
    _URC_CONTINUE_UNWIND = 8,
    _URC_FAILURE = 9   /* unspecified failure of some kind */
  }
_Unwind_Reason_Code

Any ideas?

mpromonet, 2018-03-16:
Hi  JudaismBot,
Just try and it seems to work, did you try to cross-compile for armv6 or armv7 ?
Regards,
Michel.

JudaismBot, 2018-03-19:
/gcc-linaro-6.4.1-2017.11-x86_64_arm-linux-gnueabihf

JudaismBot, 2018-03-19:
fliped around the build toolchain.  Now using a debootstaped image with QEMU and it compiled

closed 2018-03-28 by mpromonet