mpromonet/webrtc-streamer #106: Build failing in Xenial

KeMBro2012, 2018-03-21:
Trying to build under Xenial and I manage to get WebRTC and live555 built, but WebRTC-Streamer fails with the following:

`In file included from src/PeerConnectionManager.cpp:24:0:
inc/rtspvideocapturer.h: In member function ‘virtual bool RTSPAudioSource::onData(const char*, unsigned char*, ssize_t, timeval)’:
inc/rtspvideocapturer.h:181:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      while (m_buffer.size() > segmentLength*m_channel) {
                             ^
src/PeerConnectionManager.cpp: In member function ‘bool PeerConnectionManager::streamStillUsed(const string&)’:
src/PeerConnectionManager.cpp:528:29: error: ‘class webrtc::MediaStreamInterface’ has no member named ‘label’
    if (localstreams->at(i)->label() == streamLabel)
                             ^
src/PeerConnectionManager.cpp: In member function ‘const Json::Value PeerConnectionManager::hangUp(const string&)’:
src/PeerConnectionManager.cpp:557:51: error: ‘class webrtc::MediaStreamInterface’ has no member named ‘label’
    std::string streamLabel = localstreams->at(i)->label();
                                                   ^
src/PeerConnectionManager.cpp: In member function ‘const Json::Value PeerConnectionManager::getPeerConnectionList()’:
src/PeerConnectionManager.cpp:659:35: error: ‘class webrtc::MediaStreamInterface’ has no member named ‘label’
       stream[localstreams->at(i)->label()] = tracks;
                                   ^
Makefile:94: recipe for target 'src/PeerConnectionManager.o' failed
make: *** [src/PeerConnectionManager.o] Error 1`

Any help would be appreciated.

mpromonet, 2018-03-21:
Hi Keith,
This comes with the commit in webrtc [Final name changing of MediaStreamInterface.label() to id()](https://chromium.googlesource.com/external/webrtc/+/13b8bad23543c59e57428a642aa31a24f0923d76).
I will update the code to follow this change.
Best Regards,
Michel.

closed 2018-03-21 by mpromonet

KeMBro2012, 2018-03-22:
Thanks for the quick response, that did fix the issue. Now I'm back to civetweb not being able to find stdio.h, which I had figured out previously -- needed to upgrade from Trusty to Xenial (which was long overdue), re-download webrtc-streamer, and start from scratch. I'm starting from scratch again to see if maybe something got screwed up when I encountered the `label` error, but I might be back for help again if that doesn't work.

KeMBro2012, 2018-03-22:
Something happens during the build of civetweb that causes the above issue on my system; not your problem to fix.

That said, I got it fixed by cleaning up the civetweb directory and started having a different issue, so I think this needs to be reopened. To ensure that it wasn't something I did while screwing around trying to get this to build, I nuked the webrtc-streamer directory and started with a fresh clone. Now I get this on the current HEAD:

```
install -d ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
install -m 755 testMP3Streamer testMPEG1or2VideoStreamer testMPEG1or2AudioVideoStreamer testMPEG2TransportStreamer testMPEG4VideoStreamer testH264VideoStreamer testH265VideoStreamer testDVVideoStreamer testWAVAudioStreamer testAMRAudioStreamer testMKVStreamer testOggStreamer vobStreamer testMP3Receiver testMPEG1or2VideoReceiver testMPEG2TransportReceiver sapWatch testRelay testReplicator testOnDemandRTSPServer testRTSPClient openRTSP playSIP testMPEG1or2Splitter testMPEG1or2ProgramToTransportStream testH264VideoToTransportStream testH265VideoToTransportStream MPEG2TransportStreamIndexer testMPEG2TransportStreamTrickPlay registerRTSPStream ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
make[2]: Leaving directory '/home/keith/builds/webrtc-streamer/live/testProgs'
cd mediaServer ; make install
make[2]: Entering directory '/home/keith/builds/webrtc-streamer/live/mediaServer'
g++ -fno-rtti  -c -I../UsageEnvironment/include -I../groupsock/include -I../liveMedia/include -I../BasicUsageEnvironment/include -I. -O -DSOCKLEN_T=socklen_t -g -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -Wall -DBSD=1 live555MediaServer.cpp
g++ -fno-rtti  -c -I../UsageEnvironment/include -I../groupsock/include -I../liveMedia/include -I../BasicUsageEnvironment/include -I. -O -DSOCKLEN_T=socklen_t -g -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -Wall -DBSD=1 DynamicRTSPServer.cpp
g++ -olive555MediaServer -L. live555MediaServer.o DynamicRTSPServer.o ../liveMedia/libliveMedia.a ../groupsock/libgroupsock.a ../BasicUsageEnvironment/libBasicUsageEnvironment.a ../UsageEnvironment/libUsageEnvironment.a 
install -d ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
install -m 755 live555MediaServer ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
make[2]: Leaving directory '/home/keith/builds/webrtc-streamer/live/mediaServer'
cd proxyServer ; make install
make[2]: Entering directory '/home/keith/builds/webrtc-streamer/live/proxyServer'
g++ -fno-rtti  -c -I../UsageEnvironment/include -I../groupsock/include -I../liveMedia/include -I../BasicUsageEnvironment/include -I. -O -DSOCKLEN_T=socklen_t -g -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -Wall -DBSD=1 live555ProxyServer.cpp
g++ -olive555ProxyServer -L. live555ProxyServer.o ../liveMedia/libliveMedia.a ../groupsock/libgroupsock.a ../BasicUsageEnvironment/libBasicUsageEnvironment.a ../UsageEnvironment/libUsageEnvironment.a 
install -d ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
install -m 755 live555ProxyServer ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/bin
make[2]: Leaving directory '/home/keith/builds/webrtc-streamer/live/proxyServer'
make[1]: Leaving directory '/home/keith/builds/webrtc-streamer/live'
rm -rf live
```

This is after a successful build of webrtc and the live555 target.

KeMBro2012, 2018-03-22:
My mistake, I pasted the wrong output. This is the error:

```
git submodule update live555helper
make -C live555helper CC=g++ PREFIX=/usr CFLAGS_EXTRA="-Wall -pthread -g -std=c++11 -Iinc --sysroot=../webrtc/src/build/linux/debian_sid_amd64-sysroot  -DWEBRTC_POSIX -fno-rtti -DHAVE_JPEG -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include -I ../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DHAVE_ALSA -I ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include -DHAVE_LIVE555 -I live555helper/inc -I ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/liveMedia  -I ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/groupsock -I ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/UsageEnvironment -I ../webrtc/src/build/linux/debian_sid_amd64-sysroot/usr/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream" SYSROOT=../webrtc/src/build/linux/debian_sid_amd64-sysroot
make[1]: Entering directory '/home/keith/builds/webrtc-streamer/live555helper'
Makefile:13: *** Cannot find live555.  Stop.
make[1]: Leaving directory '/home/keith/builds/webrtc-streamer/live555helper'
Makefile:54: recipe for target 'live555helper/live555helper.a' failed
make: *** [live555helper/live555helper.a] Error 2
```

mpromonet, 2018-03-22:
Hi,
This is because you installed live555 in webrtc sysroot and look for it in /usr.
You should try specifying `SYSROOT=$(pwd)/../webrtc/src/build/linux/debian_sid_amd64-sysroot` as an absolute path.
The dockerfile show the defferent step of the build https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile.
Best Regards,
Michel.

KeMBro2012, 2018-03-22:
Beautiful, it appears to be building now. I'll let you know when it completes.

In the mean time you might want to update the README.md as it specifically mentions that SYSROOT defaults to ../webrtc

KeMBro2012, 2018-03-22:
Ah, I spoke too soon, now I'm getting the following:

```
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:30: undefined reference to `Json::Value::Value(Json::ValueType)'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:55: undefined reference to `Json::Reader::Reader()'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:56: undefined reference to `Json::Reader::parse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value&, bool)'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:66: undefined reference to `Json::Value::isNull() const'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:68: undefined reference to `Json::StyledWriter::StyledWriter()'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:68: undefined reference to `Json::StyledWriter::write[abi:cxx11](Json::Value const&)'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:63: undefined reference to `Json::Value::~Value()'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:30: undefined reference to `Json::Value::~Value()'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:63: undefined reference to `Json::Value::~Value()'
/home/keith/builds/webrtc-streamer/src/HttpServerRequestHandler.cpp:30: undefined reference to `Json::Value::~Value()'
collect2: error: ld returned 1 exit status
Makefile:98: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1
```

Pages and pages of similar messages.

mpromonet, 2018-03-22:
Hi,
It seems you missed the options `use_custom_libcxx=false` running `gn`
Best Regards,
Michel.

KeMBro2012, 2018-03-22:
Here is the gn command line I used:
```
gn gen out/Release --args='is_debug=false use_custom_libcxx=false rtc_use_h264=true ffmpeg_branding="Chrome" rtc_include_tests=false use_ozone=true rtc_include_pulse_audio=false rtc_build_examples=false'
```

`use_custom_libcxx=false` is present there; this is the line from your documentation, plus `rtc_build_examples=false`, so there's something else going on, I think.

mpromonet, 2018-03-22:
Hi,
You can check the symbols in the libwebrtc*.a running something like :+1: 
`nm libwebrtc*.a | c++filt | grep Json::Value`
If symbols are not preset you should remove the libwebrtc*.a and rebuild it.
Regards,
Michel.

KeMBro2012, 2018-03-23:
huh... I don't have any files anywhere in (including subdirectories) my webrtc/ webrtc-streamer/ directories...

Alright, time to start digging to find out where things are going wrong! I'm fairly certain both webrtc and live555 are building correctly.

ushalin, 2018-04-03:
@KeMBro2012 Were you able to find a solution to this problem?

KeMBro2012, 2018-04-03:
I haven't had a chance to come back to this since I last commented. I'm
hoping to give it another go over the weekend and will update when I do.

commit 456667de (2018-03-21): fix #106: update with new webrtc https://chromium.googlesource.com/external/webrtc/+/13b8bad23543c59e57428a642aa31a24f0923d76