mpromonet/webrtc-streamer #109: Build Issues

ushalin, 2018-04-03:
Hey there! Are there certain versions of WebRTC that webrtc-streamer doesn't build with it? I'm currently trying to build with the latest versions of both, and get issues when I make. Specfically

`src/PeerConnectionManager.cpp:14:60: fatal error: api/audio_codecs/builtin_audio_encoder_factory.h: No such file or directory`

Any help would be great, thanks!

mpromonet, 2018-04-03:
Hi Shalin,
In the latest webrtc the file seems to exists [builtin_audio_encoder_factory.h](https://chromium.googlesource.com/external/webrtc/+/master/api/audio_codecs/builtin_audio_encoder_factory.h).
I try to maintain build with latest webrtc.
The last CI build use [webrtc commit 3d954a6](https://chromium.googlesource.com/external/webrtc/+/3d954a6962e5084d428f61b36bf189a9f9c9b0e9)
Best Regards,
Michel.

mpromonet, 2018-04-04:
Hi,
In fact the default for WEBRTCROOT is not ../webrtc, but $(pwd)/../webrtc.
As you are using the default value, did you try simply : 

      make

Best Regards,
Michel.

ushalin, 2018-04-04:
Hey,
I did a git pull on WebRTC and did a git checkout on commit `3d954a6`. Now when I make it, I get a different error:

```g++ -o src/main.o -c src/main.cpp -Wall -pthread -g -std=c++11 -Iinc --sysroot=/  -DWEBRTC_POSIX -fno-rtti -DHAVE_JPEG -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include -I ../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DHAVE_ALSA -I //usr/include -DHAVE_LIVE555 -I live555helper/inc -I //usr/include/liveMedia  -I //usr/include/groupsock -I //usr/include/UsageEnvironment -I //usr/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream -DVERSION="\"v0.0.8-dirty webrtc@3d954a6 live555helper@5d3072a civetweb@v1.10-149-g5f92d24 h264bitstream@0.2.0\""
g++ -o webrtc-streamer src/PeerConnectionManager.o src/rtspvideocapturer.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a live555helper/live555helper.a civetweb/libcivetweb.a h264bitstream/.libs/libh264bitstream.a -pthread --sysroot=/ -ldl -lrt  -L //usr/lib -lasound live555helper/live555helper.a -L //usr/lib -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a  -L civetweb -l civetweb h264bitstream/.libs/libh264bitstream.a
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::ConvertVideoEncoderFactory(std::unique_ptr<cricket::WebRtcVideoEncoderFactory, std::default_delete<cricket::WebRtcVideoEncoderFactory> >)':
convert_legacy_video_factory.cc:(.text._ZN7cricket26ConvertVideoEncoderFactoryESt10unique_ptrINS_25WebRtcVideoEncoderFactoryESt14default_deleteIS1_EE+0x43): undefined reference to `cricket::InternalEncoderFactory::InternalEncoderFactory()'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::ConvertVideoDecoderFactory(std::unique_ptr<cricket::WebRtcVideoDecoderFactory, std::default_delete<cricket::WebRtcVideoDecoderFactory> >)':
convert_legacy_video_factory.cc:(.text._ZN7cricket26ConvertVideoDecoderFactoryESt10unique_ptrINS_25WebRtcVideoDecoderFactoryESt14default_deleteIS1_EE+0x43): undefined reference to `cricket::InternalDecoderFactory::InternalDecoderFactory()'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::(anonymous namespace)::EncoderAdapter::GetSupportedFormats() const':
convert_legacy_video_factory.cc:(.text._ZNK7cricket12_GLOBAL__N_114EncoderAdapter19GetSupportedFormatsEv+0x2c): undefined reference to `cricket::InternalEncoderFactory::InternalEncoderFactory()'
convert_legacy_video_factory.cc:(.text._ZNK7cricket12_GLOBAL__N_114EncoderAdapter19GetSupportedFormatsEv+0x34): undefined reference to `cricket::InternalEncoderFactory::supported_codecs() const'
convert_legacy_video_factory.cc:(.text._ZNK7cricket12_GLOBAL__N_114EncoderAdapter19GetSupportedFormatsEv+0xd4): undefined reference to `cricket::InternalEncoderFactory::~InternalEncoderFactory()'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::(anonymous namespace)::EncoderAdapter::CreateVideoEncoder(webrtc::SdpVideoFormat const&)':
convert_legacy_video_factory.cc:(.text._ZN7cricket12_GLOBAL__N_114EncoderAdapter18CreateVideoEncoderERKN6webrtc14SdpVideoFormatE+0x89): undefined reference to `webrtc::SimulcastEncoderAdapter::SimulcastEncoderAdapter(cricket::WebRtcVideoEncoderFactory*)'
convert_legacy_video_factory.cc:(.text._ZN7cricket12_GLOBAL__N_114EncoderAdapter18CreateVideoEncoderERKN6webrtc14SdpVideoFormatE+0xd8): undefined reference to `webrtc::VideoEncoderSoftwareFallbackWrapper::VideoEncoderSoftwareFallbackWrapper(cricket::VideoCodec const&, std::unique_ptr<webrtc::VideoEncoder, std::default_delete<webrtc::VideoEncoder> >)'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::(anonymous namespace)::DecoderAdapter::CreateVideoDecoder(webrtc::SdpVideoFormat const&)':
convert_legacy_video_factory.cc:(.text._ZN7cricket12_GLOBAL__N_114DecoderAdapter18CreateVideoDecoderERKN6webrtc14SdpVideoFormatE+0xff): undefined reference to `webrtc::VideoDecoderSoftwareFallbackWrapper::VideoDecoderSoftwareFallbackWrapper(webrtc::VideoCodecType, std::unique_ptr<webrtc::VideoDecoder, std::default_delete<webrtc::VideoDecoder> >)'
libWebRTC__Release.a(vp8_encoder_simulcast_proxy.o): In function `webrtc::VP8EncoderSimulcastProxy::InitEncode(webrtc::VideoCodec const*, int, unsigned long)':
vp8_encoder_simulcast_proxy.cc:(.text._ZN6webrtc24VP8EncoderSimulcastProxy10InitEncodeEPKNS_10VideoCodecEim+0x3e): undefined reference to `webrtc::SimulcastEncoderAdapter::SimulcastEncoderAdapter(cricket::WebRtcVideoEncoderFactory*)'
collect2: error: ld returned 1 exit status
Makefile:98: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1```

ushalin, 2018-04-04:
Hey,
I did a git pull on WebRTC and a git checkout to `3d954a6`. On webrtc-streamer, I'm on commit `1d9952e`. When I run `make` I now get a new issue:

```g++ -o src/main.o -c src/main.cpp -Wall -pthread -g -std=c++11 -Iinc  --sysroot=/home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot  -DWEBRTC_POSIX -fno-rtti -DHAVE_JPEG -I /home/shalin/workspace/webrtc-streamer/../webrtc/src -I /home/shalin/workspace/webrtc-streamer/../webrtc/src/third_party/jsoncpp/source/include -I /home/shalin/workspace/webrtc-streamer/../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DUSE_X11 -DHAVE_ALSA -I /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/include -DHAVE_LIVE555 -I live555helper/inc -I /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/include/liveMedia  -I /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/include/groupsock -I /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/include/UsageEnvironment -I /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream -DVERSION="\"v0.0.9-20-g1d9952e webrtc@3d954a6 live555helper@5d3072a civetweb@v1.10-149-g5f92d24 h264bitstream@0.2.0\""
g++ -o webrtc-streamer src/PeerConnectionManager.o src/rtspvideocapturer.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a live555helper/live555helper.a civetweb/libcivetweb.a h264bitstream/.libs/libh264bitstream.a -pthread  --sysroot=/home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot -ldl -lrt  -lX11 -lXext -lXdamage -lXfixes -lXcomposite  -L /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/lib -lasound live555helper/live555helper.a -L /home/shalin/workspace/webrtc-streamer/../webrtc/src//out/Release/../../build/linux/debian_sid_amd64-sysroot/usr/lib -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a  -L civetweb -l civetweb h264bitstream/.libs/libh264bitstream.a
src/PeerConnectionManager.o: In function `ScreenCapturer::ScreenCapturer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:28: undefined reference to `webrtc::DesktopCaptureOptions::CreateDefault()'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:28: undefined reference to `webrtc::DesktopCapturer::CreateWindowCapturer(webrtc::DesktopCaptureOptions const&)'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:28: undefined reference to `webrtc::DesktopCaptureOptions::~DesktopCaptureOptions()'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:48: undefined reference to `webrtc::DesktopCaptureOptions::CreateDefault()'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:48: undefined reference to `webrtc::DesktopCapturer::CreateScreenCapturer(webrtc::DesktopCaptureOptions const&)'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:48: undefined reference to `webrtc::DesktopCaptureOptions::~DesktopCaptureOptions()'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:28: undefined reference to `webrtc::DesktopCaptureOptions::~DesktopCaptureOptions()'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:48: undefined reference to `webrtc::DesktopCaptureOptions::~DesktopCaptureOptions()'
src/PeerConnectionManager.o: In function `ScreenCapturer::OnCaptureResult(webrtc::DesktopCapturer::Result, std::unique_ptr<webrtc::DesktopFrame, std::default_delete<webrtc::DesktopFrame> >)':
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:60: undefined reference to `webrtc::DesktopFrame::rect() const'
/home/shalin/workspace/webrtc-streamer/inc/screencapturer.h:61: undefined reference to `webrtc::DesktopFrame::rect() const'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::ConvertVideoEncoderFactory(std::unique_ptr<cricket::WebRtcVideoEncoderFactory, std::default_delete<cricket::WebRtcVideoEncoderFactory> >)':
convert_legacy_video_factory.cc:(.text._ZN7cricket26ConvertVideoEncoderFactoryESt10unique_ptrINS_25WebRtcVideoEncoderFactoryESt14default_deleteIS1_EE+0x43): undefined reference to `cricket::InternalEncoderFactory::InternalEncoderFactory()'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::ConvertVideoDecoderFactory(std::unique_ptr<cricket::WebRtcVideoDecoderFactory, std::default_delete<cricket::WebRtcVideoDecoderFactory> >)':
convert_legacy_video_factory.cc:(.text._ZN7cricket26ConvertVideoDecoderFactoryESt10unique_ptrINS_25WebRtcVideoDecoderFactoryESt14default_deleteIS1_EE+0x43): undefined reference to `cricket::InternalDecoderFactory::InternalDecoderFactory()'
libWebRTC__Release.a(convert_legacy_video_factory.o): In function `cricket::(anonymous namespace)::EncoderAdapter::GetSupportedFormats() const':
convert_legacy_video_factory.cc:(.text._ZNK7cricket12_GLOBAL__N_114EncoderAdapter19GetSupportedFormatsEv+0x2c): undefined reference to `cricket::InternalEncoderFactory::InternalEncoderFactory()'
[... cut ...]
libWebRTC__Release.a(vp8_encoder_simulcast_proxy.o): In function `webrtc::VP8EncoderSimulcastProxy::InitEncode(webrtc::VideoCodec const*, int, unsigned long)':
vp8_encoder_simulcast_proxy.cc:(.text._ZN6webrtc24VP8EncoderSimulcastProxy10InitEncodeEPKNS_10VideoCodecEim+0x3e): undefined reference to `webrtc::SimulcastEncoderAdapter::SimulcastEncoderAdapter(cricket::WebRtcVideoEncoderFactory*)'
collect2: error: ld returned 1 exit status
Makefile:113: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1

ushalin, 2018-04-04:
Hey Michel,
I got it to build and run!
I did a fresh fetch of WebRTC and then followed the steps in your README, and performed a `make`, rather than a `make WEBRTCROOT=../webrtc WEBRTCBUILD=Release`.
However, I'm still not sure if the issues I faced were due to something I did wrong, or code-side issues, as I performed a `make` on my other instance of WebRTC and got the issue in my comment above.

Regardless, I'll close the issue.

Thank you for help!

closed 2018-04-04 by ushalin