mpromonet/webrtc-streamer #110: Feel really stupid...

JudaismBot, 2018-04-03:
make[1]: Entering directory `/home/hmibuild/webrtc-streamer/live555helper'
g++ -c -o src/environment.o src/environment.cpp -W -Wall -pthread -g -pipe -Wall -pthread -g -std=c++11 -Iinc  --sysroot=/home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot  -DWEBRTC_POSIX -fno-rtti -DHAVE_JPEG -I /home/hmibuild/webrtc-streamer/../webrtc/src -I /home/hmibuild/webrtc-streamer/../webrtc/src/third_party/jsoncpp/source/include -I /home/hmibuild/webrtc-streamer/../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DHAVE_LIVE555 -I live555helper/inc -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot/usr/include/liveMedia  -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot/usr/include/groupsock -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot/usr/include/UsageEnvironment -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot/usr/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream -I inc -g  -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot//usr/include/liveMedia  -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot//usr/include/groupsock -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot//usr/include/UsageEnvironment -I /home/hmibuild/webrtc-streamer/../webrtc/src/out/Release/../../../build/linux/debian_wheezy_amd64-sysroot//usr/include/BasicUsageEnvironment/
src/environment.cpp:10:20: fatal error: iostream: No such file or directory
 #include <iostream>
                    ^

Hints?

ushalin, 2018-04-03:
Were you running `make live555`? If so, try running with it with sudo, I think I had a similar issue, where `sudo make live555` fixed it.

dherman516, 2018-04-03:
This was for the primary "make" after the live555 one...
using defaults for the command line

mpromonet, 2018-04-03:
Hi,
The error message means probably the SYSROOT is wrongly computed, or not present...
By the way webrtc seems really old, we see `debian_wheezy_amd64-sysroot` in your log, it is now `debian_sid_amd64-sysroot`.
You should try to update it (using `git pull origin master` in webrtc/src, clean the src/out/Release, and rebuild it.
Best Regards,
Michel.

dherman516, 2018-04-04:
That solved it! Thanks!

closed 2018-04-04 by dherman516