mpromonet/webrtc-streamer #111: Usage Question: External Http Server

dherman516, 2018-04-04:
With the -H parameter, do you have a sample implementation guide for how to use?

I assume this is to create a separation of devices similar to the peerconnection_client/server examples?  

My use case is one in which the streamer has no open incoming ports, it has to operate and function purely from websocket commands (from an external server) and then the webrtc_streamer calls the peer

mpromonet, 2018-04-06:
Hi,
The option '-H' allows to define the interface/port to bind to. 
As it is directly set to the civetweb option you can get documentation in https://github.com/civetweb/civetweb/blob/master/docs/UserManual.md#listening_ports-8080
It could be for instance:
 
* 127.0.0.1:8080 if you like to access only from localhost
* H8000,9000 open 2 HTTP port
* -H8443s open an HTTPS port
* -H8080r,8443s redirect HTTP port 8080 to HTTPS port 8553
* ...

Best Regards,
Michel.

mpromonet, 2018-04-23:
HI Dave,
I close this issue, if my answer is not enough clear, please open a new one appending more details of what you like to do.
Best Regards,
Michel.

closed 2018-04-23 by mpromonet

neucoreAn, 2019-02-27:
HI  DaveH，
is your plan to remove the internally integrated web server and interact through the external web server?