mpromonet/webrtc-streamer #112: Gstreamer / Raspicam Source RTSP Issues

dherman516, 2018-04-04:
webrtc-streamer running off a Ubuntu Xenail server (AM64).  When I initialize a RTSP stream  (running of a raspi -- gstreamer / rpicamsrc / gst-rtsp server), I get mmal errors fail to initialize component ENOSPC. 

If I start webrtc-streamer prior to the rtsp server, then it works... Once..

This raspi set up works fine with Janus streaming test and VLC.

dherman516, 2018-04-05:
https://github.com/thaytan/gst-rpicamsrc/issues/74

The issue is that for some reason Live555 was opening up two RTSP connections (which caused a block and a crash of the RTSP server due to the camera being in use).

mpromonet, 2018-04-05:
Hi,
It open one for video, one for audio, but you can avoid this setting the audio url calling the api /call.
Best Regards,
Michel.

closed 2018-04-05 by dherman516