mpromonet/webrtc-streamer #113: Usage Question: Embed in HTML

ushalin, 2018-04-05:
When I run `./webrtc-streamer` it sends it to localhost:8000. So what do I have to do to be able to access this stream in any HTML page?

dherman516, 2018-04-05:
http://<IP Adress where webrtc-streamer is runnning>:8000

ushalin, 2018-04-05:
Sorry for the lack of detail in my initial question. I meant to say, how can I give it a specific IP address to run on, and if I already have an existing HTML file (Ex. index.html), what do I have to do in index.html to play the stream?

dherman516, 2018-04-05:
You will have to set up a target URL in the java script to point back at your webrtc-streamer server on port 8000

mpromonet, 2018-04-05:
Hi,
There is en embedded index.html, you can try it from the live demo https://rtsp2webrtc.herokuapp.com
You can also specify the url you want to display like this https://rtsp2webrtc.herokuapp.com/webrtcstreamer.html?video=rtsp://217.17.220.110/axis-media/media.amp
And  you can create your HTML using the WebRtcStreamer JS interface or use directly the HTTP API.
Best Regards,
Michel.

ushalin, 2018-04-06:
Hi Michel,
Could you explain a bit more on how its possible to host the html using another server and connect to the webrtc stream?

mpromonet, 2018-04-06:
Hi Shalin,
You can serve an HTML page that create a WebRtcStreamer specifying the webrtc-streamer url to use to get webrtc streams : 
 
         var webRtcServer      = new WebRtcStreamer(<video tag>, <webrtc-streamer url>);
Next you can connect the element with id '<video tag>' to a webrtc stream using : 

         webRtcServer.connect(<webrtc stream name>)

Best Regards,
Michel.

mpromonet, 2018-04-23:
Hi Shalin,
The `webrtc-streamer url`  is the url of the embeded http server (that serve the API), you can test it with https://rtsp2webrtc.herokuapp.com/ serving your own pages with another HTTP server.
And the `webrtc stream name` is the name of the webrtc stream that you like to open, you can "enumerate" it using /getMediaList (event this is not completly true because you can reach rtsp url that are not listed... but this is another story).
Best Regards,
Michel.

closed 2018-05-11 by mpromonet