mpromonet/webrtc-streamer #114: Building this project for RasPI via cross compile

dherman516, 2018-04-05:
OK ...  Building webrtc is done..
Building this project is a mess. 

I tried setting 
export CROSS=arm-linux-gnueabihf-
and a bunch of other export variables...

The Live 555 makefile just brute force uses g++, so it looks like short of major edits to the makefile in question, I have to compile on native

dherman516, 2018-04-05:
I build the webrtc bits on an ubuntu box... tar balled 'em all up to move to my raspi

mpromonet, 2018-04-05:
Hi,
I don't know how o give access to the circleci artefact but you can try the lastest release https://github.com/mpromonet/webrtc-streamer/releases/download/v0.0.9/webrtc-streamer_v0.0.9_rpi-armv7.tgz
Best Regards,
Michel.

dherman516, 2018-04-09:
Cool it runs.. but it appears the foced path for the alsa config is giving it fits
>ALSA lib conf.c:3942:(snd_config_update_r) Cannot access file /webrtc/src/rpi-armv7/out/Release/../../../build/linux/debian_sid_arm-sysroot/usr/share/alsa/alsa.conf

dherman516, 2018-04-09:
Also getting this error:  [19133] (stunport.cc:274): >Port[0x761176b8:video:1:0:local:Net[enxb827ebceba3f:19.7.94.x/23:Unknown]]: UDP send of 1174 bytes failed with error 11

closed 2018-04-09 by dherman516