mpromonet/webrtc-streamer #116: STUN not being honored

dherman516, 2018-04-09:
Using a command line for stun and turn (to same address)..   looking at javascript from 

[image]

dherman516, 2018-04-10:
in main.cpp  the line "if (optarg)"  will never ever trigger so, the optarg is never used

mpromonet, 2018-04-10:
Hi,
If you refer to line 180, it is trigged when -s have a argument (note that you should not have a space between -s and the argument as it is an parameter that could have an argument or not).
If you try -sdummy, the stunurl will be dummy.
Best Regards,
Michel.

dherman516, 2018-04-10:
./webrtc-streamer -s 19.86.98.249:3478  rtsp://19.7.95.146:8554/test rtsp://192.168.1.26:8554/wiredCamera

with the logging changes I pushed. Stun server is google's not the specified one

mpromonet, 2018-04-10:
As I try to explain in my previous comment, you shoule use `-s19.86.98.249:3478` without space.
In your case, -s has no value.

dherman516, 2018-04-10:
Got it.. turn seems to work with a space though

closed 2018-04-11 by dherman516

dherman516, 2018-04-11:
I am adding to my fork secure stun support.. take a look if interested... only like 10 lines of code changes

dherman516, 2018-05-14:
I also found that if you set an iptables firewall it confuses the hell out of the stun server