mpromonet/webrtc-streamer #120: Question On How To LiveStream RTP

ybchoo, 2018-04-17:
Hi There
I have a livestream from rtp://192.168.2.1/ 
But stream is in .h264 format. Can kindly show some example on how can I retrieve these file using webrtc?

mpromonet, 2018-04-20:
Hi ybchoo,
This is quite straight forward, you just need to create a live555 MediaSession from an SDP file instead of the RTSP DESCRIBE.
You can find a example of this from this [answer](https://stackoverflow.com/a/32595407/3102264)
Best Regards,
Michel.

ybchoo, 2018-04-21:
Thanks Mike
Will look into it.

mpromonet, 2018-04-23:
Hi ybchoo,
I made a part of the job adding to live555helper the capacity to connect to a stream using an SDP file in commit https://github.com/mpromonet/live555helper/commit/ecadcebb1459f62d488cee530d33c3af6c2cc7dd

In order to connect an rtp stream, you need to make the connection between a VideoCapturer and the SDPClient, an example is [testSdpClient.cpp](https://github.com/mpromonet/live555helper/blob/master/testSdpClient.cpp).
I don't know if I will do this implementation, probably nearly all cameras have now an RTSP implemenation.
Best Regards,
Michel.

ybchoo, 2018-04-24:
> to a stream using an SDP file in commit mpromonet/live555helper@ecadceb

Thanks Mick. Appreciated your clear explanation.

closed 2019-05-03 by mpromonet