mpromonet/webrtc-streamer #121: Block webcam access

ushalin, 2018-04-18:
Is it possible to block `webrtc-streamer` from every getting access to my webcam (i.e. /dev/video0)?

mpromonet, 2018-04-20:
HI Shalin,
This need a little modification in the code, or you can run under a user that cannot access to /dev/video0.
Best Regards,
Michel.

ushalin, 2018-04-20:
Do you know where in the code base the modification would be?

mpromonet, 2018-04-21:
Hi Shalin,
I made a little [modification](https://github.com/mpromonet/webrtc-streamer/commit/ea493ee18bc824542648ca0b9557b627e55e2118) that allow to configure what is enumerated by api /getMediaList.
By default everything is enumerated but you can restrict this using: 

     -q"window://" to enumerate only windows
     -q"videocap://|audiocap://" to enumerate video devices & audio devices

However if the http client know your webcam name, it still can access to it.
If you like you just need to remove the device creation in  `PeerConnectionManager::CreateVideoTrack`
Best Regards,
Michel.

mpromonet, 2018-04-21:
Hi Shalin,
I made an second modification to not allow access to what is not enumerated, then using : 

   -q"rtsp://.*|screen://.*" you will not get the list of v4l2 capture device and will not have access to it.

Best Regards,
Michel.