mpromonet/webrtc-streamer #125: ([006:507] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529)How can i pull  rtsp(h264)  from a ipcamera to play on chrome via webrtc?

dpmaycry, 2018-06-15:
I have a RTSP url form camera

and I did the follow:
1. docker pull mpromonet/webrtc-streamer
2. docker run -p 8000:8000 -it mpromonet/webrtc-streamer -n raspicam -u rtsp://url

But only get error like:
[h264 @ 0x7f60cc00cb80] no frame!
[006:507] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529

but a vlc player can show the video from camera via  rtsp.

dpmaycry, 2018-06-15:
logs are as follow:

Logger level:4
[000:000] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:000] [1] (audio_mixer_manager_alsa_linux.cc:174): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:000] [1] (adm_helpers.cc:47): Unable to access speaker.
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:001] [1] (audio_mixer_manager_alsa_linux.cc:174): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:001] [1] (audio_device_alsa_linux.cc:796): unable to open playback device: No such file or directory (-2)
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:001] [1] (audio_mixer_manager_alsa_linux.cc:254): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:001] [1] (adm_helpers.cc:67): Unable to access microphone.
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:001] [1] (audio_mixer_manager_alsa_linux.cc:254): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:001] [1] (audio_device_alsa_linux.cc:919): unable to open record device: No such file or directory
[000:001] [1] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:002] [1] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:002] [1] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
[010:808] [11] (shared_x_display.cc:38): Unable to open display
[010:808] [11] (shared_x_display.cc:38): Unable to open display
[010:829] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.5852481797924365
[010:832] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.5852481797924365
Created new TCP socket 79 for connection
Created new TCP socket 83 for connection
[010:834] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.5852481797924365
[010:835] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.5852481797924365
Start playing sink for "video/H264" subsession
markerSize:4
[h264 @ 0x7f225400cc40] no frame!
[010:902] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[012:945] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[014:952] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[016:961] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[018:964] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[019:269] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.326170234791024
[019:272] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.326170234791024
[019:274] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.326170234791024
[019:276] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.326170234791024
[h264 @ 0x7f225400cc40] no frame!
[020:972] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[022:976] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[h264 @ 0x7f225400cc40] no frame!
[024:983] [66] (h264_decoder_impl.cc:330): avcodec_send_packet error: -1094995529
[025:238] [11] (PeerConnectionManager.cpp:526): Close PeerConnection
[025:240] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:5 peerid:0.5852481797924365
[025:245] [8] (sctptransport.cc:823): SctpTransport->SendQueuedStreamResets(): Failed to send a stream reset for 1 streams : [0x00000009] Bad file descriptor
[026:342] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.20131138199539866
[... cut ...]
[043:826] [1] (PeerConnectionManager.h:205): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.011518805639319618
[043:833] [1] (PeerConnectionManager.h:211): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:5 peerid:0.011518805639319618
[043:836] [8] (sctptransport.cc:823): SctpTransport->SendQueuedStreamResets(): Failed to send a stream reset for 1 streams : [0x00000009] Bad file descriptor

mpromonet, 2018-06-16:
Hi Dante,
Did you try to increase the log level to see if something could explain why the stream is not decoded ?
Best Regards,
Michel

dpmaycry, 2018-06-16:
Hi Michel: 
Thanks for reply,I 'll try it  next week(new to docker ...).
My best guess is Chrome did't decode h264 with ffmpeg but v8/v9
I am now using electron to develop a realtime rtsp video client with your streamer( Hope that the delay will be less than 200~300ms in local network ).
Your Sincerely,
Dante

mpromonet, 2018-06-16:
Hi Dante,
The video format between the camera and the RTSP capturer is usually different from the format between the browser and the WebRTC source. The backend could use YUYV with a webcam, or MJPEG/H264 with the RTSP capturer, and the frontend could use VP8.   
The log about avcodec_send_packet error is often due to non-video frames (AUD, SEI,...) and may not a be problem...
Best Regards,
Michel.

dpmaycry, 2018-06-19:
Chrome on my mac does the trick...
but Chrome on win cannot
all using lasted version of chrome...

closed 2018-09-27 by mpromonet