mpromonet/webrtc-streamer #127: Compile failure on x86_64 since CircleCl build #397

jcarterof, 2018-06-21:
I tried a clean build today on Ubuntu 18.04 and I see the same failure in the webrtc build that I see on the CircleCl builds from the last few days, beginning with 397.  I found that if I remove  "use_custom_libcxx=false" from the gn args that webrtc will build just fine.  Of course, the subsequent steps to build webrtc-streamer then fail though.

mpromonet, 2018-06-23:
Hi jcarterof,
It seems to build with 'is_clang=false', if the build goes to the end, I will commit this option that is already used for arm build.
Best Regards,
Michel.

jcarterof, 2018-06-25:
Hi Michel,
I heard back from the webrtc team on [issue 9438](https://bugs.chromium.org/p/webrtc/issues/detail?id=9438).  It seems that in addition to "use_custom_libcxx=false" we need "use_custom_libcxx_for_host=true".  I just ran a test build of webrtc only and it worked.
John

jcarterof, 2018-06-25:
Hi Michel,
I tried the full webrtc-streamer build using the additional GN arg and now I see errors on the final link.  It seems that the use_custom_libcxx_for_host=true arg allows webrtc to build just fine but causes other issues for us downstream.

John

jcarterof, 2018-06-25:
Hi Michel,
I see now that you got it working in circleci.  I have incorporated the GN args that you used there into my build and everything works now.

Many thanks for the prompt response.

John

closed 2018-06-30 by mpromonet