mpromonet/webrtc-streamer #128: Strange behavior with external webserver

paoValle, 2018-06-29:
Hi, i am trying to restream rtsp IP CAM video to webrtc. I have tried the docker Solution and the "vanilla" one, downloading the last release of your software. It works if i start the software and after connect to the webrtc video through the page on civet webserver. But, if i start the software and i try to see the video through a page made on my local apache server, using the Simple index in your readme, it's like the video is not available. If the i THEN open the page in civet, it won't start anymore. The strange things is that if i first start the software, THEN open the page on civet and THEN on my apache, it works. Maybe it's like the simple page doesn't start something? Am i doing something wrong? My need is to start the streamer through console commando and THEN Access to the video from my own web page. Is it possibile?
To recap the situation:
1) start docker container, open The page on civet IP, open The page on my local apache: WORK
2) start docker container, open The page on civet, open The page on local apache, close the page on civet, reload the page on local apache: DONT WORK.
3) start docker container, open local apache page: DONT WORK.
has the page on civet to be open? Are some socket or releated closed or not open for some reason?
I hope i made Clear the situation

mpromonet, 2018-07-15:
Hi Paolo,
One thing that is different on the simple page in the README.md and the page index.html served by the embeded HTTP server is the option for the rtsp connection.  
The embdeded page use options defined in https://github.com/mpromonet/webrtc-streamer-html/blob/819831a8ff92ec5e413bb012ee8b49d4d1d8c6a0/webrtcstreamer.json, so I thing you sould have the same behaviour using : 

    window.onload         = function() { webRtcServer.connect("rtsp://pi2.local:8554/unicast","","rtptransport=tcp&timeout=60") }

Once the backend stream is created, the options given to further connect are ignored.
Maybe It could be beeter to not shared the conenction that are using different option...
Best Regards,
Michel.

paoValle, 2018-07-15:
for some reason, firefox doesn't always create the ice connection, while chrome always can. 

Now i have another problem. I'm trying a new rtsp stream from another camera. if i try it on vlc it works, while if i try it on your software it crashes, giving a message about RTP payload malformed or not valid. Really strange. Maybe the camera that is streaming the rtsp creates the rtp package with some strange format and your software fails during parsing? what do you think?

mpromonet, 2018-07-16:
Hi Poalo,
vlc & webrtc-streamer use both live555 as rtsp client, so it should work more or less the same.
Without more logs or investigation I cannot say more.
By the way this problem seems really different from the original issue, it is better to not mix the problem in the same issue.
Best Regards,
Michel.

tarikziyad9, 2018-08-02:
Hi paoValle,

could you share with us how you connect the docker image to Apache server, I'm trying to do it but it isn't working

closed 2018-08-14 by mpromonet