mpromonet/webrtc-streamer #132: RTP Payload makes webrtc streamer to crash, while vlc works

paoValle, 2018-07-16:
Hi, it's strange, i have 3 vivotek cameras and i am trying to use this software to push video to browser via webrtc. The cameras stream via RTSP, obv. I can make your streamer to work with 2 of them, while 1 camera causes the software to crash (seg. fault). This is the error message:

https://pastebin.com/eVnzJemQ

The strange fact is that vlc has no problem to play the stream. Looking at debug of vlc and webrtcstreamer, i've noted that vlc uses live555 version from 2014, while webrtcstreamer uses 2018 version. Maybe a bug/wrong parser added with the new versions?

What can you tell me?

mpromonet, 2018-07-16:
Hi Paolo,
This is open and free software distributed under [unlicense](https://github.com/mpromonet/webrtc-streamer/blob/master/LICENSE) , I could advise you to run a debugger and make some investigation, this could start with some "thread apply all where".
Feel free to submit a pull request to fix the problem you meet to share with other users.
Best Regards,
Michel.

paoValle, 2018-07-16:
i would be glad to give an hand to your work, but i have no idea where to start from. Can you give me some advice or some king of inspection to do?

what do you think if i try to compile webrtc with the same version of live555 of vlc? could it work?

mpromonet, 2018-07-17:
Hi Paolo,

In your log we see that the process crash, then the first thing to do, is to understand where.

To know this, the simplest way is to run the program using gdb with something like : 
   gdb --ex run --ex "thread apply all where" --batch  --args ./webrtc-streamer -v -n mystream -u rtsp://something

When the crash occurs, it should print the stack of every threads and exit of debugger.
Best Regards,
Michel.

closed 2018-10-18 by mpromonet