mpromonet/webrtc-streamer #134: Network is unreachable

MichaelB2018, 2018-07-27:
I get lots of 

[18023:121] [7516] (stunport.cc:529): sendto : [0x00000065] Network is unreachable

and my video is not stable. Any idea what I'm doing wrong?

MichaelB2018, 2018-07-28:
Hi Michel,

Thanks for the response. Will you push out a new release?

PS: I also get this error message:
   **[066:364] [15768] (stunport.cc:87): Binding request timed out from [0:0:0:x:x:x:x:x]:34794 (any)**
Is this related too?

Michael

mpromonet, 2018-07-28:
Hi Michael,
The logic is in WebRTC code, in order to workaroud this I disable IPV6 it at the operating system level.
Best Regards,
Michel.

MichaelB2018, 2018-07-29:
Oh, and sometimes after I disconnect, I get a segmentation fault:

**[532:563] [21772] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:closing
Segmentation fault**

Any way I can help with more information on what causes the segmentation fault? Would a core dump help? Or maybe increase log level?

mpromonet, 2018-07-29:
Hi Michael,
You could post the backtrace to understand what cause the SEGV.
Best Regards,
Michel

MichaelB2018, 2018-07-30:
Hi Michel
I used your pre-compiled version. Do I need to recompile it myself to enable backtraces? Or is there any other way?
Michael

closed 2018-07-30 by mpromonet