mpromonet/webrtc-streamer #135: instruction to run the docker with external webserver

tarikziyad9, 2018-08-02:
I want to run the streamer using external Tomcat HTTP server running in the localhost at port 8082,

I write the followings instruction:

```
run -p 8000:8000 -it mpromonet/webrtc-streamer -n raspicam -u rtsp://184.72.239.149/vod/mp4:Bi
gBuckBunny_115k.mov -H 127.0.0.1:8082
```

The docker says that: 
HTTP Listen at 127.0.0.1:8082

But when I'm opening this URL it guides me to the Tomcat management page.
 
could you provide me please the correct instruction needed for that? will the streamer be deployed automatically in the Tomcat using the instruction?

mpromonet, 2018-08-02:
Hi tarikziyad9,
You are making some confusion, using -H inside the docker container is probably not what you like to do. But I did not understand what you like to do, even if the page is served by the tomcat server, the page need to access to the streamer to get the webrtc stream.

The docker file expose the port 8000 of the container that is the default port of the streamer.
What you could do is to map the internal port 8000 to an other one.
For instance running : 

docker run -p 9000:8000 -it mpromonet/webrtc-streamer -n raspicam -u rtsp://184.72.239.149/vod/mp4:BigBuckBunny_115k.mov

Will allow to access to the streamer on port 9000.

Then If you like to serve an html page from tomcat that access to the streamer, you could inspire from the example in [README.md](README.md)

```
<html>
<head>
<script src="request.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
    var webRtcServer      = new WebRtcStreamer("video",location.protocol+"//"+window.location.hostname+":9000");
    window.onload         = function() { webRtcServer.connect("raspicam") }
    window.onbeforeunload = function() { webRtcServer.disconnect() }
</script>
</head>
<body> 
    <video id="video" />
</body>
</html>

```

Best Regards,
Michel

tarikziyad9, 2018-08-03:
Hi Michel,

Great thanks to you, it works perfectly!

But the HTML page you wrote didn't respond, so I requested the RTSP URL through the index page as

`http://localhost:9000/?video=rtsp://184.72.239.149/vod/mp4:BigBuckBunny_115k.mov`
 
and I changed the webrtcstreamer.json to be as

`var webrtcConfig = {
	url: "http://localhost:9000",
	options: "rtptransport=tcp&timeout=60"
}
`

and It worked fine.

Best Regards,
Tarik

closed 2018-08-14 by mpromonet