mpromonet/webrtc-streamer #136: In versions that are ported to Windows, program is killed when browser(Chrome) closed

sacoku, 2018-08-14:
Hi. all

I had ported with **windows version** in last-week.

This time, I doing stress test. 
by the way, i found few bug.
This bug(below) may only occur on Windows

- when browser close, program is killed(connection fail signal handling)

I think that this bug occur in PeerConnectionManager::hangup.
In the Code, PeerConnectionObserver called PeerConnectionManager::hangup.
and PeerConnectionManager::hangup method try to delete PeerConnectionObserver
I think that method calling structure is ambiguous.

mpromonet, 2018-08-14:
Hi Kim,
This doesnot cause problem still this is the last call in the hangup mathod, the object will just no longer exists. You could also remove this call to check your hypothesis.
Looking to the code, the `peer_connectionobs_map_` is not protected by a mutex, this could brings problems, if different thread are using it.
Best Regards,
Michel.

sacoku, 2018-08-16:
Thanks for your reply.

sacoku, 2018-08-16:
Thank for your reply.
I solve this issue.(below)
1. Added sychronization(mutex) peer_connectionobs_map_
2. deleted PeerConnectionObserver object in another thread

I don’t know that this way is correct. but solved this issue(program killed in windows environment. When browser close.

closed 2018-08-16 by sacoku

mpromonet, 2018-08-21:
Hi Kim,
Maybe you can share your work ?
Best Regards,
Michel.