mpromonet/webrtc-streamer #138: I want to know why my rtsp stream is not work ? thanks!!!

461316334, 2018-08-22:
this is the test log \:

```log
Logger level:3
[000:000] [1] (audio_device_impl.cc:249): PulseAudio is disabled using build flag.
[000:001] [1] (audio_device_alsa_linux.cc:156): failed to open X display, typing detection will not work
[000:001] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
[000:002] [1] (audio_mixer_manager_alsa_linux.cc:174): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:002] [1] (adm_helpers.cc:47): Unable to access speaker.
[000:002] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
[000:002] [1] (audio_mixer_manager_alsa_linux.cc:174): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:002] [1] (audio_device_alsa_linux.cc:759): InitSpeaker() failed
[000:002] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM
[000:002] [1] (audio_device_alsa_linux.cc:796): unable to open playback device: No such file or directory (-2)
[000:003] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
[000:003] [1] (audio_mixer_manager_alsa_linux.cc:254): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:003] [1] (adm_helpers.cc:67): Unable to access microphone.
[000:003] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
[000:003] [1] (audio_mixer_manager_alsa_linux.cc:254): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:003] [1] (audio_device_alsa_linux.cc:880): InitMicrophone() failed
[000:003] [1] (audio_device_alsa_linux.cc:1348): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM
[000:003] [1] (audio_device_alsa_linux.cc:919): unable to open record device: No such file or directory
[000:003] [1] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:003] [1] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:003] [1] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
[006:688] [1] (PeerConnectionManager.h:209): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3951039038476556
[006:690] [1] (PeerConnectionManager.h:215): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.3951039038476556
[006:692] [1] (PeerConnectionManager.h:209): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3951039038476556
[006:693] [1] (PeerConnectionManager.h:215): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.3951039038476556
[006:695] [8] (physicalsocketserver.cc:562): Socket::OPT_DSCP not supported.
[006:696] [8] (stunport.cc:427): STUN server address is incompatible.
[006:697] [8] (thread.cc:271): Waiting for the thread to join, but blocking calls have been disallowed
[006:697] [8] (thread.cc:271): Waiting for the thread to join, but blocking calls have been disallowed
[006:698] [8] (stunport.cc:400): Port[0x7f817003daa0:audio:1:0:local:Net[any:0:0:0:x:x:x:x:x/0:Wildcard:id=0]]: StunPort: stun host lookup received error 0
[006:698] [8] (thread.cc:271): Waiting for the thread to join, but blocking calls have been disallowed
[006:796] [8] (physicalsocketserver.cc:562): Socket::OPT_DSCP not supported.
[006:818] [8] (jseptransportcontroller.cc:303): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
[006:824] [8] (jseptransportcontroller.cc:303): Not adding candidate because the JsepTransport doesn't exist. Ignore it.
Created new TCP socket 80 for connection
Connecting to 47.99.77.140, port 17970 on socket 80...
Created new TCP socket 83 for connection
Connecting to 47.99.77.140, port 17970 on socket 83...
...remote connection opened
Sending request: DESCRIBE rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2018.08.05
Accept: application/sdp

...remote connection opened
Sending request: DESCRIBE rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2018.08.05
Accept: application/sdp

Received 184 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 401 Unauthorized
CSeq: 2
WWW-Authenticate: Digest realm="IP Camera(C2189)", nonce="29d283a6e8d8c33e98dda250e7f8e74c", stale="FALSE"
Date:  Wed, Aug 22 2018 09:27:02 GMT

Resending...
Sending request: DESCRIBE rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream RTSP/1.0
[... 199 lines cut ...]
Start playing sink for "video/H264" subsession
markerSize:4
Sending request: PLAY rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream/ RTSP/1.0
CSeq: 5
Authorization: Digest username="admin", realm="IP Camera(C2189)", nonce="35e912482266275f1b76729fdc65969e", uri="rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream/", response="2ed48d34d2fa02520be73682701123c8"
User-Agent: LIVE555 Streaming Media v2018.08.05
Session: 1862257664
Range: npt=0.000-

Received 220 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 5
Session:       1862257664
RTP-Info: url=rtsp://admin:admin000@21m677n127.imwork.net:17970/h264/ch1/main/av_stream/trackID=1;seq=53632;rtptime=378035294
Date:  Wed, Aug 22 2018 09:27:14 GMT

[021:846] [1] (PeerConnectionManager.cpp:531): Close PeerConnection
[021:846] [1] (PeerConnectionManager.cpp:544): Close PeerConnection no more used rtspadminadmin000@21m677n127imworknet17970h264ch1mainav_stream|
[021:846] [1] (PeerConnectionManager.cpp:556): Close PeerConnection removed rtspadminadmin000@21m677n127imworknet17970h264ch1mainav_stream|
[021:846] [1] (PeerConnectionManager.h:209): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3951039038476556
Close session: video/H264
[021:851] [1] (PeerConnectionManager.h:209): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3951039038476556
[021:861] [1] (PeerConnectionManager.h:215): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:5 peerid:0.3951039038476556
```

461316334, 2018-08-22:
I can use vlc-player to play it!

mpromonet, 2018-08-22:
Hi 461316334,
Did you try to use RTP over TCP instead of UDP (vls automatically switch if nothing is received on UDP) ?
You can also try to give an url for audio to avoid the second connection.
Increasing the loglevel can also helps you to understand the issue...
Best Regards,
Michel.

mpromonet, 2018-08-22:
Hi 461316334,
In order to use RTP over tcp, you could set `rtptransport=tcp` in option  [WebRtcStreamer.connect](http://htmlpreview.github.io/?https://github.com/mpromonet/webrtc-streamer/blob/master/jsdoc/WebRtcStreamer.html)
Best Regards,
Michel.

461316334, 2018-08-23:
thanks a lot , it works !!! 
Best Regards just for you.

closed 2018-08-23 by 461316334