mpromonet/webrtc-streamer #139: Can you help integrate python-openCV to this

daljeet-hcl, 2018-08-26:
Hi there,
 your project is very interesting, I have a question, can you help me please

I read frames an from python bindings to openCV, then do some processing on the frame, I want to send these frames to WebRTC as video with minimum Latency, How can I do it using your webrtc Streamer.

Any hints?

mpromonet, 2018-08-26:
Hi daljeet-hcl,

I think a possible way is to use a v4l2loopback device, using opencv you can write to it and next webrtc-streamer could read from it.

Best Regards,

Michel.

daljeet-hcl, 2018-08-26:
Thanks Michel, for thequick answer, I will definitely try that.

 Is there a better and direct way to write frames directly  ? so that the latency is further reduced.

Regards
Daljeet

mpromonet, 2018-08-27:
Hi Daljeet,

If you like to avoid the v4l2loopback, an other approach could be using a other capturer, for instance using zeromq. There is a fork https://github.com/abhikandoi/webrtc-streamer that experiment this way.
One advantage using zeromq is to avoid kernel module, after for performance, it needs to make some tests....

Best Regards,

Michel.

daljeet-hcl, 2018-08-29:
I decided to go with the v4l2loopback and created a device named "/dev/video1"
the following commands which read data from my webCam and write to /dev/video1 worked with webrtc-streamer : 

> 1. gst-launch-1.0 v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1 
> 2. ffmpeg -re -i /dev/video0 -f v4l2 /dev/video1 

However when I write to /dev/Video1 using openCV and ffmpeg, I am able to read the /dev/video1 in VLC but it doe not work with webrtc-streamer : 

> python3 videoWriter.py |  ffmpeg -f rawvideo -pix_fmt bgr24 -s 640x480 -r 30 -i - -f v4l2 /dev/video1

Here is my python code of videoWriter:

https://gist.github.com/daljeet-hcl/cd55af7607c475d6f77731dc63e185dd

daljeet-hcl, 2018-08-30:
I got a quick solution to my problem, I created two v4l2 loopback devices 
modprobe v4l2loopback devices=2

then
python3 videoWriter.py | ffmpeg -f rawvideo -pix_fmt bgr24 -s 640x480 -r 30 -i - -f v4l2 /dev/video2

redirected output of device2 to device1 :
gst-launch-1.0 v4l2src device="/dev/video2" ! v4l2sink device="/dev/video1"

then was able to /dev/video1 sucessfully with webrtc-streamer.

daljeet-hcl, 2018-08-30:
Also I found that  https://github.com/jremmons/pyfakewebcam works well with webrc-streamer.

I was able to writer frames from opencv usng pyfakewebcam to v4l2Loopback device and stream it using webrtc-streamer.

https://gist.github.com/daljeet-hcl/9eee797715ac504e1694846d65eb3f78

mpromonet, 2018-08-30:
Hi daljeet-hcl,
Maybe piping is not a good idea because v4l2 is expected to read a frame in one read.
Did you rty to write to v4l2 device in your python code ?
Best Regards,
Michel.

daljeet-hcl, 2018-08-31:
https://github.com/jremmons/pyfakewebcam this project is working fine for me, thanks for such a great and easy to configure project. Let me know if I can help.
I am leaning python, but I know Java and a little nodejs.

dmathew93, 2019-02-14:
Hi, 

Can you elaborate more on how you were able to use pyfakewebcam to hook with webrtc-streamer?