mpromonet/webrtc-streamer #147: Cross-compile to RPi libx11 problem

LanderN, 2018-09-06:
Hi!
When cross-compiling for the Raspberry Pi 3, the `make all tgz` step fails with the following error messages:
```
/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lX11
/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lXext
/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lXdamage
/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lXfixes
/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lXcomposite
```

It seems X11 is not included in the toolchain? How would I fix this problem?

Thanks!

mpromonet, 2018-09-06:
Hi Lander,
You can remove these dependencies compiling WebRTC with `use_ozone=true`  like for build  `build_rpi_armv7` in https://github.com/mpromonet/webrtc-streamer/blob/master/.circleci/config.yml.
Best Regards,
Michel.

LanderN, 2018-09-07:
Thank you! This fixed my problem.

closed 2018-09-07 by LanderN