mpromonet/webrtc-streamer #150: Api hangs after call to unreachable rtsp stream

surovtsev, 2018-09-24:
First of all thank you for your work! Webrtc-streamer is really great and easy to use software. 

We are using docker image mpromonet/webrtc-streamer:release-0.0.12 (same results with latest)
to show video from ip-cam in browser.
After call api 

`/api/call?peerid=0.9817156945287386&url=rtsp%3A%2F%2Fadmin%3admin%40172.23.31.171%3A554%2F1%2F2`

with unreachable host 172.23.31.171 

other valid api calls to /api/call finish unsuccessful (timeout with 30 seconds) 

logs:

```
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection
Connecting to 172.23.31.171, port 554 on socket 80...
Created new TCP socket 80 for connection

very long log

```

How to set a attempt count to unreachable host?
Why webrtc-streamer stops respond to other requests?

mpromonet, 2018-09-24:
Hi Alexander,
I made some debugging yesterday on this subject, there is a deadlock closing the peerconnection, still now I did not understand if the problem is in WebRTC or in this project.... to workaround this I made an awful asynchronous stuff. I should find a better way.
It should be better after commit https://github.com/mpromonet/webrtc-streamer/commit/e77834a7371c84d33a10d93e309341e6053d590b
Best Regards,
Michel.

surovtsev, 2018-09-25:
Thank you, for your commit!

I updated docker container to latest version (image 234a243d7b38a43379a7babb7ad4030a092e5cb7f5ba3b8750782d1a00f9e67f).

Now Api successfully response, but still many records in log about unsuccessful connections retries to unreachable host:

Created new TCP socket 195 for connection
Connecting to 172.23.31.147, port 554 on socket 195...
Created new TCP socket 202 for connection
Connecting to 172.23.31.147, port 554 on socket 202...
Created new TCP socket 206 for connection
Connecting to 172.23.31.147, port 554 on socket 206...
Created new TCP socket 71 for connection
Connecting to 172.23.31.147, port 554 on socket 71...
Created new TCP socket 82 for connection
Connecting to 172.23.31.147, port 554 on socket 82...
Created new TCP socket 75 for connection
Connecting to 172.23.31.147, port 554 on socket 75...
Created new TCP socket 194 for connection
Connecting to 172.23.31.147, port 554 on socket 194...
Created new TCP socket 73 for connection
Connecting to 172.23.31.147, port 554 on socket 73...
Created new TCP socket 72 for connection
Connecting to 172.23.31.147, port 554 on socket 72...

mpromonet, 2018-09-27:
Hi Alexander,
I close the issue concerning "api hang".
The logs comes from underlayer, the `connecting...` should not be displayed if you doesnot add '-v' option, but the `created...` is displayed by live555 and need some overload in live555helper.
Best Regards,
Michel.

closed 2018-09-27 by mpromonet