mpromonet/webrtc-streamer #151: Multiple hangup api calls from JS not removing all streams

Kalixor, 2018-09-25:
Hello,

Awesome job this web streamer ! Thank you !
Actually, after launching several candidates peer connections (more than 7 above three permanent RTSP streams), when I refresh the web page or destroy it some streams are not removed (verifying it with /api/getStreamList).

I think this issue relies on the previous one, where it seems a thread deadlocking may occur during the 'mlock(m_streamMapMutex)' (hangup function).

When it happens (once in three), the webrtcstreamer server remains high CPU usage ans no more responsiveness, force to restart...

Thank you for the feed back.

mpromonet, 2018-09-25:
Hi Kalixo,
The hangup method is quite complex to debug as it could be called from http or from iceerror (to cleanup), still now I did not clearly understand the threading model of WebRTC SDK.
I guess the deadlock you expereriment is not related on m_streamMapMutex otherwise the getStreamList will not answer, probably more to the m_peerMapMutex. 
I guess the /getPeerConnectionList doesnot answer in such a situation (in order to avoid classical mutex order deadlock, it could be better to use only one)
Last hang was in the peerconnection->Close that was called from the peerconnection workerthread and wait for networkthread.
Using a debugger you may identify where is hang the hangup processing.
Best Regards,
Michel.

Kalixor, 2018-09-25:
Michel,

Thank you for highlights, I see, you mean this could be due to network await (like a socket in TIME_WAIT state or a lower awaiting RTSP command ...), Very probable ! Because the same JS client close correctly all streams from a different RTSP media source (where streams are heartbeated), in the JS client , I stop the heartbeats stream just before disconnecting related webstreamers , the remove streams process works fine (what ever the number of candidates, until 32).

Clearly, I need to debug where the hang occurs / and may confirms a protocol network awaiting/blocking. Since it works with heartbeated streams source I assume the `peerConnection->RemoveStream(stream);` do the job and  the `void OnRemoveStream(` is triggered...

But before I need to build it on windows...

I'll give feed back, thank you

Kalixor, 2018-09-25:
Its ok with the last build. Streams are correctly shut down few seconds (about 3 sec) after app peerconnections hang up. Thanks, Great job !

mpromonet, 2018-09-25:
Hi Kalixor,
The hang situation that was removed was in  SynchronousMethodCall::Invoke

   e_->Wait(rtc::Event::kForever);

I think the actual solution creating a temporary thread is a workaround, I will see hiw the WebRTC code evolve to find a better.
Best Regards,
Michel.

closed 2018-09-25 by mpromonet