mpromonet/webrtc-streamer #152: Build on windows

Kalixor, 2018-09-25:
I need to build a version including the last commit e77834a (for which i didn't found a release yet in the CI)
I tried to build it unsuccessfully.

Is there is a complete windows build procedure & environment settings ? (including for dependencies (webrtc / live555). I followed the commands in the readme.md, / docker file, but blocking with webrtc build (gn / ninja python commands) before launching the Cmake...

if someone succeeded on the global windows build. Thank you for any help/tips,

mpromonet, 2018-09-25:
Hi Kalixor,

The build information are in the [appveyor.yml](https://github.com/mpromonet/webrtc-streamer/blob/master/appveyor.yml).
In summary, if you fetch webrtc in ../webrtc, you should be able to build with:

    cmake -G "Visual Studio 15 2017 Win64"  . 
    msbuild INSTALL.vcxproj /p:Configuration=Release /p:Platform=x64

The cmake command will launch the `gn` and `ninja` commands, download live555 and build it inside live555helper lib.

Best Regards,
Michel.

Kalixor, 2018-09-25:
Ok perfect ....

Kalixor, 2018-09-25:
Thank you Michel, this last build works ! Previous issue : streams are correctly shut down few seconds (about 3 sec) after app peerconnections hang up.

closed 2018-09-25 by mpromonet