mpromonet/webrtc-streamer #153: Cannot Connect to Jitsi Meet Conference: /api/call fails

abhijitnathwani, 2018-09-27:
Hi @mpromonet ,

Thank you for the wonderful project. This would be helpful a lot if it I could get it to work for me. I have been following the steps and trying to connect to the Jitsi Meet room with the example you provided. I have build `webrtc-streamer` and running and providing it's argument in the `new XMPPVideoRoom("<jitsi-meet->", "<webrtc-streamer-ip>:8000");`

I am not able to connect. The /api/call/ fails. 
`10.115.3.28:8000/api/call?peerid=c0ikfktlh705c&url=undefined&audio=undefined&options=rtptransport%3Dtcp%26timeout%3D60`
If you see `&url=undefined&audio=undefined` , I don't know why it is filling it up as `undefined`. The `url` variable correctly has the `rtsp url`.

Can you please shed some light here.

Thanks,
Abhijit

mpromonet, 2018-09-27:
Hi Abhijit,

I should update the sample, the url parameter in `XMPPVideoRoom::join` should be a Object { video, audio }.
You should try something like : 

    xmpp.join("testroom", {video:"rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov"},"Bunny")

I should improve this, there is some work also to make work properly the XMPP messaging...
 
Best Regards,
Michel.

abhijitnathwani, 2018-09-27:
Hi,
Thank you for the prompt reply. I tried adding the `video` and `audio`. Still, it is failing.

API call:
`10.115.3.28:8000/api/call?peerid=4k76ssmo6vppf&url=rtsp%3A%2F%2F184.72.239.149%2Fvod%2Fmp4%3ABigBuckBunny_175k.mov&audio=rtsp%3A%2F%2F184.72.239.149%2Fvod%2Fmp4%3ABigBuckBunny_175k.mov&options=rtptransport%3Dtcp%26timeout%3D60`
Console Logs:
`
request.min.js:12 Uncaught DOMException: Failed to execute 'open' on 'XMLHttpRequest': Invalid URL
at https://10.115.3.28/jitsi/libs/request.min.js:2:1541
at tryCallTwo (https://10.115.3.28/jitsi/libs/request.min.js:10:219)
at doResolve (https://10.115.3.28/jitsi/libs/request.min.js:10:1866)
at new Promise (https://10.115.3.28/jitsi/libs/request.min.js:10:492)
at doRequest (https://10.115.3.28/jitsi/libs/request.min.js:2:48)
at XMPPVideoRoom.OnJingle (https://10.115.3.28/jitsi/xmppvideoroom.js:287:5)
at a.Handler.run (https://10.115.3.28/jitsi/libs/strophe.min.js:31:21648)
at https://10.115.3.28/jitsi/libs/strophe.min.js:31:29947
at Object.forEachChild (https://10.115.3.28/jitsi/libs/strophe.min.js:31:13463)
at a.Connection._dataRecv (https://10.115.3.28/jitsi/libs/strophe.min.js:31:29780)`

mpromonet, 2018-09-27:
Hi Abhijit,
It seems you gave a wrong url to join jitsi.
The url to join jitsi is compute in: 
 
	var XMPPVideoRoom = function XMPPVideoRoom (xmppUrl, srvurl, bus) {	
	...
		this.connection = new Strophe.Connection("https://" + this.xmppUrl + "/http-bind");
    ...
	};

Best Regards,
Michel.

abhijitnathwani, 2018-09-27:
Hi,

The Strophe Connection is succeeded and the `xmppUrl` is also proper.

Console logs:
`
Strophe is connected.
xmppvideoroom.js:384 OnPresence resource:fb122428 jid:fb122428-d339-4ee6-a1b7-23359e02dd58@10.115.3.2/81235e2e-9a85-48a2-8b16-4454a834ec4d role:moderator affiliation:owner nick:
xmppvideoroom.js:384 OnPresence resource:focus jid:focus@auth.10.115.3.2/focus276151168897126 role:moderator affiliation:owner
xmppvideoroom.js:384 OnPresence resource:Bunny jid:91d409e6-936b-406c-ba68-26db9b940909@10.115.3.2/17c2afe2-3aa2-4fef-9409-6915f0cd6217 role:participant affiliation:none status:110 nick:Bunny
`
Jitsi is successfully running on `10.115.3.2`. I also see `Bunny` as a participant on the conference, but it does not have any video and disconnects after a while.

abhijitnathwani, 2018-09-27:
Full Console Logs:

`
Strophe is connecting.
xmppvideoroom.js:68 Strophe is connected.
xmppvideoroom.js:384 OnPresence resource:fb122428 jid:fb122428-d339-4ee6-a1b7-23359e02dd58@10.115.3.2/81235e2e-9a85-48a2-8b16-4454a834ec4d role:moderator affiliation:owner nick:
xmppvideoroom.js:384 OnPresence resource:focus jid:focus@auth.10.115.3.2/focus276151168897126 role:moderator affiliation:owner
xmppvideoroom.js:384 OnPresence resource:Bunny jid:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 role:participant affiliation:none status:110 nick:Bunny
2xmppvideoroom.js:384 OnPresence resource:focus jid:focus@auth.10.115.3.2/focus276151168897126 role:moderator affiliation:owner
xmppvideoroom.js:259 OnJingle from:testroom1@conference.10.115.3.2/focus to:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 sid:57j27rqfnkc9d action:session-initiate
xmppvideoroom.js:272 <=== xmpp offer sid:57j27rqfnkc9d resource:focus initiator:focus@auth.10.115.3.2/focus276151168897126
xmppvideoroom.js:286 Method: 10.115.3.28:8000/api/call?peerid=57j27rqfnkc9d&url=rtsp%3A%2F%2F184.72.239.149%2Fvod%2Fmp4%3ABigBuckBunny_175k.mov&audio=rtsp%3A%2F%2F184.72.239.149%2Fvod%2Fmp4%3ABigBuckBunny_175k.mov&options=rtptransport%3Dtcp%26timeout%3D60
request.min.js:12 Uncaught DOMException: Failed to execute 'open' on 'XMLHttpRequest': Invalid URL
    at https://10.115.3.28/jitsi/libs/request.min.js:2:1541
    at tryCallTwo (https://10.115.3.28/jitsi/libs/request.min.js:10:219)
    at doResolve (https://10.115.3.28/jitsi/libs/request.min.js:10:1866)
    at new Promise (https://10.115.3.28/jitsi/libs/request.min.js:10:492)
    at doRequest (https://10.115.3.28/jitsi/libs/request.min.js:2:48)
    at XMPPVideoRoom.OnJingle (https://10.115.3.28/jitsi/xmppvideoroom.js:287:5)
    at a.Handler.run (https://10.115.3.28/jitsi/libs/strophe.min.js:31:21648)
    at https://10.115.3.28/jitsi/libs/strophe.min.js:31:29947
    at Object.forEachChild (https://10.115.3.28/jitsi/libs/strophe.min.js:31:13463)
    at a.Connection._dataRecv (https://10.115.3.28/jitsi/libs/strophe.min.js:31:29780)
(anonymous) @ request.min.js:12
setTimeout (async)
(anonymous) @ request.min.js:12
tryCallOne @ request.min.js:10
(anonymous) @ request.min.js:10
flush @ request.min.js:27
characterData (async)
(anonymous) @ request.min.js:27
rawAsap @ request.min.js:27
handleResolved @ request.min.js:10
handle @ request.min.js:10
Promise.then @ request.min.js:10
Promise.done @ request.min.js:12
XMPPVideoRoom.OnJingle @ xmppvideoroom.js:287
run @ strophe.min.js:31
(anonymous) @ strophe.min.js:31
forEachChild @ strophe.min.js:31
_dataRecv @ strophe.min.js:31
_onRequestStateChange @ strophe.min.js:32
XMLHttpRequest.send (async)
u @ strophe.min.js:32
_processRequest @ strophe.min.js:32
_throttledRequestHandler @ strophe.min.js:32
_onIdle @ strophe.min.js:32
_onIdle @ strophe.min.js:32
(anonymous) @ strophe.min.js:32
setTimeout (async)
_send @ strophe.min.js:32
send @ strophe.min.js:31
_onDiscoInfo @ strophe.disco.min.js:7
run @ strophe.min.js:31
(anonymous) @ strophe.min.js:31
forEachChild @ strophe.min.js:31
_dataRecv @ strophe.min.js:31
_onRequestStateChange @ strophe.min.js:32
XMLHttpRequest.send (async)
u @ strophe.min.js:32
_processRequest @ strophe.min.js:32
_throttledRequestHandler @ strophe.min.js:32
_onIdle @ strophe.min.js:32
_onIdle @ strophe.min.js:32
(anonymous) @ strophe.min.js:32
setTimeout (async)
_onIdle @ strophe.min.js:32
(anonymous) @ strophe.min.js:32
setTimeout (async)
_send @ strophe.min.js:32
send @ strophe.min.js:31
join @ strophe.muc.min.js:7
getConnection.then @ xmppvideoroom.js:45
Promise.then (async)
XMPPVideoRoom.join @ xmppvideoroom.js:28
(anonymous) @ demo.html:13
xmppvideoroom.js:259 OnJingle from:testroom1@conference.10.115.3.2/fb122428 to:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 sid:6e4c2f39bf92 action:session-initiate
xmppvideoroom.js:272 <=== xmpp offer sid:6e4c2f39bf92 resource:fb122428 initiator:testroom1@conference.10.115.3.2/fb122428
xmppvideoroom.js:259 OnJingle from:testroom1@conference.10.115.3.2/fb122428 to:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 sid:6e4c2f39bf92 action:transport-info
xmppvideoroom.js:301 <=== xmpp candidate sid:6e4c2f39bf92
xmppvideoroom.js:259 OnJingle from:testroom1@conference.10.115.3.2/fb122428 to:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 sid:6e4c2f39bf92 action:transport-info
xmppvideoroom.js:301 <=== xmpp candidate sid:6e4c2f39bf92
xmppvideoroom.js:259 OnJingle from:testroom1@conference.10.115.3.2/fb122428 to:0aa6ddb4-c87c-4bbc-9aef-c26df02c7481@10.115.3.2/b17a684b-79ea-4aa2-9bc4-abb339fdd8f0 sid:6e4c2f39bf92 action:transport-info
xmppvideoroom.js:301 <=== xmpp candidate sid:6e4c2f39bf92
`

mpromonet, 2018-09-27:
Abhijit,
It seems something is missing in the url 

> Method: 10.115.3.28:8000/api/call?

Did you try to put something before like "http://" or "https://" ? 
Regards,
Michel.

abhijitnathwani, 2018-09-27:
I added that log to identify the API call it is making. It is my custom log message added in `xmppvideoroom.js`.

abhijitnathwani, 2018-09-27:
Hi also noticed that after these logs,
I'm getting a `504 Gateway Timeout` for `https://10.115.3.2/http-bind` whereas I can clearly access that on the browser. On the browser I have `It Works! ` message. Don't know why it is getting timed out.

sminodonte, 2020-04-15:
same problem

mpromonet, 2020-04-15:
Hi,
I did not work on the jitsi stuff since a while.
If you are interest, your pull request are welcome.
Best Regards,
Michel.

mpromonet, 2020-12-17:
Hi,

I close this after fixing some problems of the jitis connection.

Best Regards,
Michel.

closed 2020-12-17 by mpromonet