mpromonet/webrtc-streamer #155: Does this support RTMP Streams?

abhijitnathwani, 2018-10-04:
Hi @mpromonet 

Thanks for this wonderful project. I was able to setup RTSP Streams with Jitsi Conference. Is there any way to include RTMP streams as well?

Thanks,
Abhijit Nathwani

mpromonet, 2018-10-08:
Hi Abhijit,

In order to connect an RTMP stream, you need to implement a VideoCapturer that support RMTP.
Maybe you can inspired from the [RTSPVideoCapturer](https://github.com/mpromonet/webrtc-streamer/blob/master/src/rtspvideocapturer.cpp)

Best Regards,
Michel.

mpromonet, 2018-10-18:
Hi Tejas,

You are right liv555 doesnot support rmtp, you should find a library that implement an rmtp client.
Probably there is such a things in ffmpeg, vlc or ...

Best Regards,
Michel

digger18, 2018-10-30:
I am interested in RTMP support as well.

mpromonet, 2018-10-31:
Hi @digger18 

Feel free to submit pull request that implement RTMP.

Best Regards,

Michel.

closed 2019-05-03 by mpromonet