mpromonet/webrtc-streamer #157: Compile for aarch64

Man777, 2018-10-07:
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::RTSPAudioSource(rtc::scoped_refptr<webrtc::AudioDecoderFactory>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:22: Warnung: undefinierter Verweis auf »rtc::Thread::Thread()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:21: Warnung: undefinierter Verweis auf »rtc::LogMessage::GetLogToDebug()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:23: Warnung: undefinierter Verweis auf »rtc::Thread::SetName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void const*)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:24: Warnung: undefinierter Verweis auf »rtc::Thread::Start(rtc::Runnable*)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:22: Warnung: undefinierter Verweis auf »rtc::Thread::~Thread()«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::~RTSPAudioSource()«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:29: Warnung: undefinierter Verweis auf »rtc::Thread::Stop()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:27: Warnung: undefinierter Verweis auf »rtc::Thread::~Thread()«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::onNewSession(char const*, char const*, char const*, char const*)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::onNewSession(char const*, char const*, char const*, char const*)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:64: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::SdpAudioFormat(absl::string_view, int, unsigned long)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:64: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:69: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::SdpAudioFormat(absl::string_view, int, unsigned long)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:69: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:74: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::SdpAudioFormat(absl::string_view, int, unsigned long)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:74: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:64: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:69: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:74: Warnung: undefinierter Verweis auf »webrtc::SdpAudioFormat::~SdpAudioFormat()«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::onData(char const*, unsigned char*, long, timeval)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/src/rtspaudiocapturer.cpp:88: Warnung: undefinierter Verweis auf »webrtc::AudioDecoder::Decode(unsigned char const*, unsigned long, int, unsigned long, short*, webrtc::AudioDecoder::SpeechType*)«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::onData(char const*, unsigned char*, long, timeval)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
src/rtspaudiocapturer.o: In Funktion »std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cricket::ToStringIfSet<bool>(char const*, absl::optional<bool> const&)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/media/base/mediachannel.h:71: Warnung: undefinierter Verweis auf »rtc::ToString[abi:cxx11](bool)«
src/rtspaudiocapturer.o: In Funktion »std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cricket::ToStringIfSet<int>(char const*, absl::optional<int> const&)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/media/base/mediachannel.h:71: Warnung: undefinierter Verweis auf »rtc::ToString[abi:cxx11](int)«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::AddSink(webrtc::AudioTrackSinkInterface*)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
src/rtspaudiocapturer.o: In Funktion »RTSPAudioSource::RemoveSink(webrtc::AudioTrackSinkInterface*)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:311: Warnung: undefinierter Verweis auf »rtc::webrtc_logging_impl::Log(rtc::webrtc_logging_impl::LogArgType const*, ...)«
src/rtspaudiocapturer.o: In Funktion »bool rtc::LogMessage::Loggable<(rtc::LoggingSeverity)2>()«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:418: Warnung: undefinierter Verweis auf »rtc::LogMessage::Loggable(rtc::LoggingSeverity)«
src/rtspaudiocapturer.o: In Funktion »bool rtc::LogMessage::Loggable<(rtc::LoggingSeverity)1>()«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:418: Warnung: undefinierter Verweis auf »rtc::LogMessage::Loggable(rtc::LoggingSeverity)«
src/rtspaudiocapturer.o: In Funktion »bool rtc::LogMessage::Loggable<(rtc::LoggingSeverity)4>()«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/rtc_base/logging.h:418: Warnung: undefinierter Verweis auf »rtc::LogMessage::Loggable(rtc::LoggingSeverity)«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0xe0): Warnung: undefinierter Verweis auf »rtc::MessageQueue::Quit()«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0xe8): Warnung: undefinierter Verweis auf »rtc::MessageQueue::IsQuitting()«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0xf0): Warnung: undefinierter Verweis auf »rtc::MessageQueue::Restart()«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0xf8): Warnung: undefinierter Verweis auf »rtc::MessageQueue::IsProcessingMessages()«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0x100): Warnung: undefinierter Verweis auf »rtc::MessageQueue::Get(rtc::Message*, int, bool)«
src/rtspaudiocapturer.o:(.data.rel.ro._ZTV15RTSPAudioSource[_ZTV15RTSPAudioSource]+0x108): Warnung: undefinierter Verweis auf »rtc::MessageQueue::Peek(rtc::Message*, int)«
[... cut ...]
src/main.o: In Funktion »std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cricket::ToStringIfSet<int>(char const*, absl::optional<int> const&)«:
/home/ubuntunt/Taye/MAN/ModelTruck/webrtc-streamer/../webrtc/src/media/base/mediachannel.h:71: Warnung: undefinierter Verweis auf »rtc::ToString[abi:cxx11](int)«
collect2: error: ld returned 1 exit status
Makefile:98: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1
ubuntunt@ubuntunt-ThinkPad-X220-Table

Man777, 2018-10-08:
Got now worked out. For x64 and aarch64 Plattform. I missed to disable the custom libcxx. 
But for other who will tries. Follow the comand down. 
This combination work for camera capture, which build in with the prozessor. 
gcc version 7.3.0 (Ubuntu/Linaro 7.3.0-27ubuntu1~18.04)

For ARM64 Architecutre, Linux

git clone --depth 1 https://chromium.googlesource.com/chromium/tools/depot_tools.git
export PATH=$PATH:$(pwd)/depot_tools
mkdir webrtc
pushd webrtc or cd webrtc
GYP_DEFINES="target_arch=arm64" fetch --no-history webrtc
gclient sync
cd src
./build/install-build-deb.sh
build/linux/sysroot_scripts/install-sysroot.py --arch=arm64
gn gen out/arm64 --args='dcheck_always_on=true ffmpeg_branding="Chrome" is_component_build=false is_debug=false rtc_use_h264=true rtc_include_tests=false use_custom_libcxx=false rtc_include_pulse_audio=false target_cpu="arm64"'
ninja -C out/arm64

sudo apt-get install gcc-aarch64-linux-gnu 
webrtc-streamer
cd webrtc-streamer
gedit Makefile -> change it WEBRTCBUILD: arm64
make CROSS=aarch64-linux-gnu- SYSROOT=/home/../webrtc/src/build/linux/debian_sid_arm64-sysroot live555
make CROSS=aarch64-linux-gnu- SYSROOT=/home/../webrtc/src/build/linux/debian_sid_arm64-sysroot tgz
->If sometimes stuck7stop, correct the Makefile, which the source is not founding. 
In my case was running the whole comile at once. Only a link error accour. But forget to disable the custom standard c++ lib.

closed 2018-10-08 by Man777

Jishang, 2019-02-01:
> Got now worked out. For x64 and aarch64 Plattform. I missed to disable the custom libcxx.
> But for other who will tries. Follow the comand down.
> This combination work for camera capture, which build [...]

I am doing the same procedure describe above still getting unknown error.
root/usr/lib -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a  -L civetweb -l civetweb

libwebrtc__arm64.a: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
Makefile:97: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1

If you have compiled or release version of that can you please provide me i want to check for my camera?
GCC version:gcc-linaro-5.5.0-2017.10-x86_64_aarch64-linux-gnu

lukelu0520, 2020-07-27:
There is no ./build/install-build-deb.sh, but ./build/install-build-deps.sh
Is it a typo ?