mpromonet/webrtc-streamer #158: Support for Firefox for Jitsi Meet Connection

abhijitnathwani, 2018-10-09:
Hi @mpromonet ,

I have noticed the Jitsi Room connection does not work on the Firefox browser. Initially I thought it is the CORS issue as per console log. I have added the necessary headers in the `nginx` config and things work fine on Chrome. Do you have any clue as to how to fix this for Firefox?

[image]

antonsimola, 2019-02-06:
Did you manage to get this to work? I'm having some issues with CORS as well.

abhijitnathwani, 2019-02-07:
Hi @antonsimola ,

Nope, I still haven't managed to get this working. Let me know if you can get this worked.

Abhijit

mpromonet, 2019-02-07:
Hi,
Usually I use https://chrome.google.com/webstore/detail/allow-control-allow-origi/nlfbmbojpeacfghkpbjhddihlkkiljbi extensin to disable CORS checking.
Best Regards,
Michel

abhijitnathwani, 2019-02-08:
Hi,

Things work well on Chrome, after using this extension or adding the CORS checking in server config. It is Firefox that is creating the problem.

Abhijit

closed 2020-12-17 by mpromonet