mpromonet/webrtc-streamer #159: Cannot send rtsp stream to janus videoroom plugin

orbroot, 2018-10-11:
Hi,

I am trying to send video from Raspberry Pi camera to a Janus videoroom plugin. To that end, I downloaded the release "webrtc-streamer_v0.1.0-dirty_rpi-armv7.tgz"

1 - launch it with the command line : 
sudo ./webrtc-streamer -vv -s janus.xxxxxxxx:3478 -t xx:xxxx@janus.xxxxxxxxxxxxxxx:3478

2 - create a RTSP tream using gst-rtsp-server
./test-launch -p 8888 "( rpicamsrc preview=false bitrate=500000 keyframe-interval=5 ! video/x-h264, width=1280, height=720, framerate=25/1 ! h264parse ! rtph264pay name=pay0 pt=96 )"

3 - join janus room with the nodejs script included
sudo node joinjanusvideoroom.js http://localhost:8000 rtsp://127.0.0.1:8888/test

I have complaints on the webrtc-streamer side, saying that there is no videourl/audiourl :
Actually, it seems not to take into account the RTSP stream passed in parameter....

---------------------------------
[125:410] [21597] (PeerConnectionManager.cpp:751): videourl:undefined
[125:410] [21597] (device_info_linux.cc:40): NumberOfDevices
[125:411] [21597] (device_info_linux.cc:65): GetDeviceName
[125:412] [21597] (webrtcvideocapturer.cc:155): Failed to find capturer for id: 0
[125:412] [21597] (PeerConnectionManager.cpp:763): Cannot create capturer video:undefined
[125:413] [21597] (PeerConnectionManager.cpp:792): audiourl:undefined
[125:413] [21597] (audio_device_impl.cc:332): Init
[125:414] [21597] (audio_device_impl.cc:727): RecordingDevices
[125:418] [21594] (opensslcertificate.cc:105): Returning certificate
[125:425] [22303] (paced_sender.cc:94): PacedSender paused.
[125:442] [21597] (audio_device_impl.cc:730): output: 0
[125:443] [21597] (PeerConnectionManager.cpp:831): audiourl:undefined idx_audioDevice:-1
[125:443] [21597] (PeerConnectionManager.cpp:842): Cannot create capturer audio:undefined
---------------------------------

Any luck you have already encoutered this issue ?

Best regards,
Orbroot

mpromonet, 2018-10-12:
Hi orbroot,
This is due to a modification that supposed url have a structure {video:"....", audio:"...."}.
I commit https://github.com/mpromonet/webrtc-streamer/commit/5d0fde2f545dc91ad2fb7b7c7ebe272aa82efab1 that use url as video url when url.video is undefined.
Best Regards,
Michel.

orbroot, 2018-10-17:
> I commit 5d0fde2

Hi,

Thank you for your kind response.

At this point, I managed to create a RTSP server, and send video data to 
it. However, when I want to join a janus room using the produced URL of 
the RTSP stream, the webrtc-server only produces hundreds of lines :

---------------------------

Created new TCP socket 85 for connection
Failed to SETUP: Missing or bad "Session:" header
[115:524] [31653] (rtspvideocapturer.cpp:207): RTSPVideoCapturer:onError 
url:rtsp://janus.xxxxxxxxxxxxxx:0000/device error:Missing or bad 
"Session:" header
Failed to PLAY: No RTSP session is currently in progress
--------------------------------

The RTSP stream is playing fine with ffmpeg.

Any ideas why it complains about missing headers ? Are those headers to 
be generated by the RTSP server ?

Best regards,

Orbroot

closed 2019-03-10 by mpromonet