mpromonet/webrtc-streamer #160: Stream local file.

Dooque, 2018-10-11:
Hi @mpromonet. Thanks for this excelente project!

I'm trying to stream a local file, at the beginning I thought it was not possible since the command line help don't say anything, but later I found into the source code that it is possible to pass an URL like `file://...`, so I tested it but nothing happened. I tried with `h264/mp4`, `h264/mpg`, and `h264/avi` just in case, in all cases I got the same output on the command line and nothing is stream into the webpage:

```bash
./webrtc-streamer -H 0.0.0.0:7000 -n video1.mpg -u file:///home/pi/video.mpg
Logger level:4
[000:000] [13347] (audio_mixer_manager_alsa_linux.cc:882): Could not find capture volume on the mixer.
[000:000] [13347] (adm_helpers.cc:67): Unable to access microphone.
[000:014] [13347] (audio_mixer_manager_alsa_linux.cc:882): Could not find capture volume on the mixer.
[000:027] [13347] (audio_device_alsa_linux.cc:924): unable to open record device: No such file or directory
[000:028] [13347] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:028] [13347] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:028] [13347] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
ALSA lib conf.c:3942:(snd_config_update_r) Cannot access file /opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../arm-linux-gnueabihf/libc/usr/share/alsa/alsa.conf
HTTP Listen at 0.0.0.0:7000
blue[014:902] [13347] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.42795455138639493
[014:918] [13347] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.42795455138639493
ERROR: File does not begin with an EBML header
[014:953] [13352] (PeerConnectionManager.cpp:831): audiourl:video1.mpg idx_audioDevice:-1
[014:953] [13352] (PeerConnectionManager.cpp:842): Cannot create capturer audio:video1.mpg
[014:953] [13347] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.42795455138639493
[014:960] [13347] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.42795455138639493
[015:065] [13347] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[015:071] [13347] (PeerConnectionManager.h:212): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[015:072] [13347] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
[015:072] [13347] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ClientDataChannel msg:local channel openned
[015:072] [13347] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ServerDataChannel msg:remote channel openned
```

I'm running the binary [webrtc-streamer_v0.1.0-dirty_rpi-armv7](https://github.com/mpromonet/webrtc-streamer/releases/download/v0.1.0/webrtc-streamer_v0.1.0-dirty_rpi-armv7.tgz) on a **Raspberry Pi 3** with a **Raspbian**.

I also try to stream the file through RTSP, using the [live555 mediaServer](http://www.live555.com/liveMedia/) and the [ffmpeg ffserver](https://www.ffmpeg.org/ffserver.html), and connect **webrtc-streamer** to that stream, but again nothing happened. This is the output (it is the same for both servers):

> Note: I can reproduce these streams with VLCPlayer.

```bash
./webrtc-streamer -H 0.0.0.0:7000 -n video.mpg -u rtsp://192.168.80.130:7575/video
Logger level:4
[000:000] [12900] (audio_mixer_manager_alsa_linux.cc:882): Could not find capture volume on the mixer.
[000:000] [12900] (adm_helpers.cc:67): Unable to access microphone.
[000:013] [12900] (audio_mixer_manager_alsa_linux.cc:882): Could not find capture volume on the mixer.
[000:027] [12900] (audio_device_alsa_linux.cc:924): unable to open record device: No such file or directory
[000:028] [12900] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:028] [12900] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:028] [12900] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
ALSA lib conf.c:3942:(snd_config_update_r) Cannot access file /opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../arm-linux-gnueabihf/libc/usr/share/alsa/alsa.conf
HTTP Listen at 0.0.0.0:7000
[006:191] [12900] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7916740518490046
[006:206] [12900] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.7916740518490046
Created new TCP socket 82 for connection
[006:228] [12905] (PeerConnectionManager.cpp:831): audiourl:video.mpg idx_audioDevice:-1
[006:228] [12905] (PeerConnectionManager.cpp:842): Cannot create capturer audio:video.mpg
[006:228] [12900] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7916740518490046
[006:232] [12900] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.7916740518490046
[010:678] [12900] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[010:684] [12900] (PeerConnectionManager.h:212): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[010:685] [12900] (PeerConnectionManager.h:133): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
[010:719] [12900] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ClientDataChannel msg:local channel openned
[010:719] [12900] (PeerConnectionManager.h:140): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer&) channel:ServerDataChannel msg:remote channel openned
```

I tested all of these combinations in an Ubuntu 16.04 *(x86_64)* with the [docker](https://hub.docker.com/r/mpromonet/webrtc-streamer/) version as well.

So, I'm stuck at this point and I have a couple of questions:
1. Is it possible to stream directly from a file?
2. Does the file need to be in a particular format/encoding?
3. As far as I can understand, streaming the file through RTSP and connecting webrtc-streamer to that stream should work. Is that correct? (I really don't like this scenario, I'm pretty sure that must be possible to stream directly from a file).

Thank you in advance for your time! Any kind of help will be very appreciated.

Dooque, 2018-10-12:
Excellent! Thank you!

yaowangvrviu, 2019-05-03:
@Dooque can you stream the local file?
I am on Win10 OS.
1. I generated a mkv file (test.mkv) with ffmpeg, video is h264, audio is opus;
2. I copy test.mkv at the same folder with webrtc-streamer.exe;
3. in browser, I input: htttp://ip:8000/webrtcstreamer.html?file://test.mkv
4. the server log shows:
[003:252] [19320] (internal_decoder_factory.cc:56): Trying to create decoder for unsupported format
and then the webrtc-streamer.exe crashed.