mpromonet/webrtc-streamer #164: Docker build fail on Ubuntu 16.04.

Dooque, 2018-10-17:
Hi. I'm trying to build the Docker image on an Ubuntu 16.04 and it fails when executes the `cmake .` command because it cannot find the `civetweb/src/civetweb.c` file.

Here is the console output:

```bash
$ git clone https://github.com/mpromonet/webrtc-streamer.git
Cloning into 'webrtc-streamer'...
remote: Enumerating objects: 82, done.
remote: Counting objects: 100% (82/82), done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 3450 (delta 42), reused 46 (delta 21), pack-reused 3368
Receiving objects: 100% (3450/3450), 3.46 MiB | 1.44 MiB/s, done.
Resolving deltas: 100% (2251/2251), done.
Checking connectivity... done.
$ cd webrtc-streamer/
$ docker build .
Sending build context to Docker daemon  5.833MB
Step 1/10 : FROM heroku/heroku:18
18: Pulling from heroku/heroku
124c757242f8: Pull complete 
9d866f8bde2a: Pull complete 
fa3f2f277e67: Pull complete 
398d32b153e8: Pull complete 
afde35469481: Pull complete 
91b6cc55a430: Pull complete 
7ee128b37e64: Pull complete 
Digest: sha256:db39e3c25cd26a4f1a51d00a2be1d6399fcd2ae8da56142395a69a6158cf186f
Status: Downloaded newer image for heroku/heroku:18
 ---> 565f7f17c420
Step 2/10 : MAINTAINER michel.promonet@free.fr
 ---> Running in d5d550e1bc2b
Removing intermediate container d5d550e1bc2b
 ---> 1ed911f5f76e
Step 3/10 : WORKDIR /webrtc-streamer
 ---> Running in ad7233924e46
Removing intermediate container ad7233924e46
 ---> a6c2b5b7f973
Step 4/10 : COPY . /webrtc-streamer
 ---> f68642b3fac3
Step 5/10 : RUN git clone --depth 1 https://chromium.googlesource.com/chromium/tools/depot_tools.git /webrtc/depot_tools
 ---> Running in 18720ccb90b6
Cloning into '/webrtc/depot_tools'...
Removing intermediate container 18720ccb90b6
 ---> 23686124407b
Step 6/10 : ENV PATH /webrtc/depot_tools:$PATH
 ---> Running in 85cdf39fc031
Removing intermediate container 85cdf39fc031
 ---> 9179d9b945d3
Step 7/10 : RUN apt-get update && apt-get install -y --no-install-recommends g++ autoconf automake libtool xz-utils libasound2-dev libgtk-3-dev cmake p7zip-full 	&& cd /webrtc 	&& fetch --no-history --nohooks webrtc 	&& sed -i -e "s|'src/resources'],|'src/resources'],'condition':'rtc_include_tests==true',|" src/DEPS 	&& gclient sync 	&& cd /webrtc-streamer 	&& cmake . && make 	&& wget --no-check-certificate https://45.249.110.23/video/mkv/480/big_buck_bunny_480p_1mb.mkv 	&& rm -rf /webrtc && rm -f *.a && rm -f src/*.o 	&& apt-get clean && rm -rf /var/lib/apt/lists/
 ---> Running in 0991c88fd143
.
.
.

7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=C,Utf16=off,HugeFiles=on,64 bits,4 CPUs Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz (306C3),ASM,AES-NI)

Extracting archive: 
--
Path = 
Type = tar
Code Page = UTF-8

Everything is Ok

Folders: 13
Files: 496
Size:       3078657
Compressed: 261632
-- Configuring done
CMake Error at CMakeLists.txt:41 (add_library):
  Cannot find source file:

    civetweb/src/civetweb.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: Cannot determine link language for target "civetweb".
CMake Error: CMake can not determine linker language for target: civetweb
-- Generating done
-- Build files have been written to: /webrtc-streamer

The command '/bin/sh -c apt-get update && apt-get install -y --no-install-recommends g++ autoconf automake libtool xz-utils libasound2-dev libgtk-3-dev cmake p7zip-full 	&& cd /webrtc 	&& fetch --no-history --nohooks webrtc 	&& sed -i -e "s|'src/resources'],|'src/resources'],'condition':'rtc_include_tests==true',|" src/DEPS 	&& gclient sync 	&& cd /webrtc-streamer 	&& cmake . && make 	&& wget --no-check-certificate https://45.249.110.23/video/mkv/480/big_buck_bunny_480p_1mb.mkv 	&& rm -rf /webrtc && rm -f *.a && rm -f src/*.o 	&& apt-get clean && rm -rf /var/lib/apt/lists/' returned a non-zero code: 1
$
```

Rakiah, 2018-10-17:
Did you pulled the submodules before running the dockerfile ?
```
git submodule update --init
```

Dooque, 2018-10-17:
That's the problem! My bad! Thanks!

closed 2018-10-17 by Dooque

Bhlowe, 2019-02-06:
I added this to the Dockerfile, right before cmake. Any reason this shouldn't be added?

        && git submodule update --init \

I like the idea of clone & docker build .