mpromonet/webrtc-streamer #169: BasicUsageEnvironment.hh: No such file or directory

peebles, 2018-10-28:
I've installed the Rpi toolchain, then obtained and built WebRTC as instructed in your wiki.  I then did the following:

```sh
git clone https://github.com/mpromonet/webrtc-streamer
cd webrtc-streamer/
git submodule update --init
cmake . -DWEBRTCROOT=/work
make
```

And it fails:

```sh
[ 22%] Building CXX object CMakeFiles/live555helper.dir/live555helper/src/sdpclient.cpp.o
In file included from /work/webrtc-streamer/live555helper/inc/sdpclient.h:14:0,
                 from /work/webrtc-streamer/live555helper/src/sdpclient.cpp:13:
/work/webrtc-streamer/live555helper/inc/environment.h:12:36: fatal error: BasicUsageEnvironment.hh: No such file or directory
```

I cannot find a "BasicUsageEnvironment.hh" anywhere.

??

peebles, 2018-10-28:
Following a hint from another issue, I found that if I did the following:

```sh
wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz
tar zxf live555-latest.tar.gz
cmake
make
```

It gets past all of the live555 build steps, but then fails here:

```sh
[ 97%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/main.cpp.o
/work/webrtc-streamer/src/main.cpp:13:33: fatal error: rtc_base/ssladapter.h: No such file or directory
compilation terminated.
```

So close ...

mpromonet, 2018-10-28:
Hi peebles,

You are right WEBRTCROOT should point on src, this is confusing, I will change back to use the root directory (like it was with Makefile).

In order to not have the X11 dependency, you should build using WEBRTCOZONE=Yes.

However you are probably not cross-compiling, you are compiling for your taget.

In order to use a cross-compiler the CMakeList.txt need some modification. Still now only the Makefile can cross-compile (see https://github.com/mpromonet/webrtc-streamer/blob/master/.circleci/config.yml or https://github.com/mpromonet/webrtc-streamer/blob/master/.travis.yml)

Best Regards,
Michel.

peebles, 2018-10-28:
I want to try to get the cross compile to work, and I would like to get rid of as many dependencies as possible (esp. X11!).  From the instructions on the wiki, "gn gen" has already been run with "use_ozone=true" on its command line, yet webrtc-streamer still seems to want to include the X11 libs: 

```sh
cmake . -DWEBRTCROOT=/work/src -DWEBRTCOZONE=Yes
make

[100%] Linking CXX executable webrtc-streamer
/usr/bin/ld: cannot find -lX11
/usr/bin/ld: cannot find -lXext
/usr/bin/ld: cannot find -lXdamage
/usr/bin/ld: cannot find -lXfixes
/usr/bin/ld: cannot find -lXcomposite
collect2: error: ld returned 1 exit status
```

And if I try make without cmake:

```sh
make clean
cmake clean .
git checkout Makefile (so I have the original now)
# make CROSS=arm-linux-gnueabihf- SYSROOT=$(pwd)/../src/build/linux/debian_sid_arm-sysroot WEBRTCROOT=/work  WEBRTCOZONE=Yes tgz
SYSROOT=/work/webrtc-streamer/../src/build/linux/debian_sid_arm-sysroot
nm: 'a.out': No such file
LIVE555 not found in /work/webrtc-streamer/../src/build/linux/debian_sid_arm-sysroot/usr/include
arm-linux-gnueabihf-g++ -o webrtc-streamer src/rtspaudiocapturer.o src/screencapturer.o src/PeerConnectionManager.o src/rtspvideocapturer.o src/filecapturer.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a civetweb/libcivetweb.a -pthread  --sysroot=/work/webrtc-streamer/../src/build/linux/debian_sid_arm-sysroot -ldl -lrt  -lX11 -lXext -lXdamage -lXfixes -lXcomposite  -L /work/webrtc-streamer/../src/build/linux/debian_sid_arm-sysroot/usr/lib -lasound -L civetweb -l civetweb

/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin/../lib/gcc/arm-linux-gnueabihf/6.4.1/../../../../arm-linux-gnueabihf/bin/ld: warning: libxcb.so.1, needed by /work/webrtc-streamer/../src/build/linux/debian_sid_arm-sysroot/usr/lib/arm-linux-gnueabihf/libX11.so, not found (try using -rpath or -rpath-link)
```

Note that the link line still contains the X libs.  If I try to execute the link line manually without the X libs, it obvious that the webrtc-streamer expects them to be included.

peebles, 2018-10-28:
Finally got it.  The trick is to use this command line:

```sh
make GYP_GENERATOR_OUTPUT=arm \
          CROSS=arm-linux-gnueabihf- \
          WEBRTCROOT=/work/webrtc \
          SYSROOT=/work/webrtc/src/build/linux/debian_sid_arm-sysroot tgz
```

Where GYP_GENERATOR_OUTPUT needed to match the args to ninja (-C arm/out/Release) specified in the wiki page.  WEBRTCROOT points to the top of the place where we build WebRTC (and does not include "/src"), and SYSROOT has to be set explicitly and correctly based on manual observation.

I should probably note that I am building inside of a docker environment where "FROM ubuntu:16.04".  I am targeting the Rpi3.

I also modified the ninja targets based on your .circleci/config.yml.  Figured you know best.

I have not yet actually tried the resulting binary.  Will update this issue with those results.  But in the mean time here is my Dockerfile:

```sh
# Build Raspberry Pi "webrtc-streamer" executable in a cross-compile
# environment on Ubuntu 16.04.  The result is a .tgz file that can
# be copied to the Rpi and extracted and run (see the end of this file
# for details.

FROM ubuntu:16.04
RUN apt-get update && apt-get install -y git curl wget python xz-utils bzip2 pkg-config

RUN mkdir /work
WORKDIR /work

# Get Rpi cross compile toolchain
RUN wget -qO- https://releases.linaro.org/components/toolchain/binaries/latest-6/arm-linux-gnueabihf/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf.tar.xz  | tar xJ -C /opt
ENV PATH=/opt/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf/bin:$PATH

# Get & Build WebRTC Native for Raspberry Pi
RUN git clone --depth 1 https://chromium.googlesource.com/chromium/tools/depot_tools.git
ENV PATH=$PATH:/work/depot_tools
RUN mkdir webrtc
WORKDIR /work/webrtc
RUN fetch --no-history --nohooks webrtc
RUN src/build/linux/sysroot_scripts/install-sysroot.py --arch=arm
RUN gclient sync
RUN sed -i -e "s|const auto iter = FindSourceInList(audio_source, &audio_source_list_);|const auto iter = std::find_if(audio_source_list_.begin(), audio_source_list_.end(), [audio_source](const std::unique_ptr<AudioMixerImpl::SourceStatus> \&p) { return p->audio_source == audio_source; });|" src/modules/audio_mixer/audio_mixer_impl.cc

WORKDIR /work/webrtc/src
RUN gn gen arm/out/Release --args='is_debug=false rtc_use_h264=true ffmpeg_branding="Chrome" is_clang=false target_cpu="arm" treat_warnings_as_errors=false rtc_include_tests=false rtc_enable_protobuf=false use_custom_libcxx=false use_ozone=true rtc_include_pulse_audio=false rtc_build_examples=false'
RUN ninja -C arm/out/Release webrtc rtc_json jsoncpp builtin_video_decoder_factory builtin_video_encoder_factory create_pc_factory peerconnection

# Get & Build & pkg WebRTC Streamer
WORKDIR /work
RUN git clone https://github.com/mpromonet/webrtc-streamer
WORKDIR /work/webrtc-streamer
RUN wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz
RUN tar zxf live555-latest.tar.gz
RUN git submodule update --init
RUN make GYP_GENERATOR_OUTPUT=arm CROSS=arm-linux-gnueabihf- WEBRTCROOT=/work/webrtc SYSROOT=/work/webrtc/src/build/linux/debian_sid_arm-sysroot tgz

# Copy webrtc-streamer_v0.1.0-22-g08bcc55_arm.tgz to the Rpi, extract it and run:
# ./webrtc-streamer
```

peebles, 2018-10-28:
Using strip on the binary gets me to 7.2Mb.  
I see you are playing with DataChannel in html/webrtcstreamer.js, but it looks like just console.logs.  Where/how is tis terminated in ./src/*.cpp?  I guess I'll need to do more research, and maybe actually read the specs!

mpromonet, 2018-10-28:
Hi peebles,

Makefile detect the X11 dependency looking to 
 $(WEBRTCLIBPATH)/obj/modules/desktop_capture/desktop_capture.ninja, as you build WebRTC for your native system, this file was created and then the Makefile still try to link to X11.

Anyway, you need to cross-compile also WebRTC.

Regards,
Michel.

girish1729, 2018-12-12:
I am trying to cross compile as well, if I run into issues I shall let you know. It is for an ARM chipset as well, I shall keep you posted on how it goes. Need to wrap it up soon and get the 7 mb size as @peebles  reported. Thanks.

girish1729, 2018-12-12:
It says 

Fatal error: invalid -march= option: `armv7-a'

I am compiling for arm-hisiv100-linux-uclibcgnueabi- 
 chipset.

Has this been attempted before?

mpromonet, 2018-12-12:
Hi Girish,
Your post is not related to this issue, issue is not a blog, I will remove your post.
Feel free to submit a dedicated issue for your problem.
However I am afraid the error message occurs building webrtc and then you will need either to contact webcrt support or patching the webrtc build process.
Best Regards,
Michel

closed 2018-12-24 by mpromonet