mpromonet/webrtc-streamer #173: Croaked: attempting to read more than once at the same time!

rossbcan, 2018-11-13:
Attempting to test audio layer

Play this file, from: https://www.montanalinux.org/files/vp9/2008-Big_Buck_Bunny-496k.webm
(https://www.montanalinux.org/taxonomy/term/29)
config:
...     "urls":{
        "Bunny": {"video":"file:///usr/webrtc-streamer/2008-Big_Buck_Bunny-496k.webm"}, ...

log:
HTTP Listen at 192.168.10.1:8000
[017:494] [2580] (PeerConnectionManager.h:216): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid$
[017:496] [2580] (PeerConnectionManager.h:222): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::Peer$
track:1
Start playing sink for "video/VP9" codecPrivate:(NULL)
markerSize:0
track:2
Start playing sink for "audio/OPUS" codecPrivate:T3B1c0hlYWQBAjgBgLsAAAAAAA==
markerSize:0
FramedSource[0x7f7e7c00ef00]::getNextFrame(): attempting to read more than once at the same time!

Here's a thread regarding this issue, may be of help:

https://live-devel.live555.narkive.com/Ip8Gf15Y/framesource-getnextframe-error-while-streaming-pcm-frames

Regards;
Bill

mpromonet, 2019-05-01:
Hi Bill,
The problem is quite different, you are trying to read an OPUS audio, that is not implemented.
However the problem is easily reproducable with [live555helper](https://github.com/mpromonet/live555helper)
Best Regards,
Michel.

mpromonet, 2019-08-02:
Hi Bill,
Testing with the latest live555 2019.07.27, it doesnot crash anymore.
However VP9 and OPUS are not implemented in the filecpaturer, then nothing is played.
Best Regards,
Michel.

closed 2019-08-02 by mpromonet