mpromonet/webrtc-streamer #178: Cross Compiling for Pi fails with new cross compiler

dmathew93, 2018-12-08:
I tried to follow your guide listed for the cross compiling but i cant get it to work at all 

the compiler im trying to use is gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf since the one listed in the guide doesnt exist anymore, 404'd

mpromonet, 2018-12-08:
Hi,

In the circleci script I use actually : 
https://releases.linaro.org/components/toolchain/binaries/6.4-2018.05/arm-linux-gnueabihf/gcc-linaro-6.4.1-2018.05-x86_64_arm-linux-gnueabihf.tar.xz 

Best Regards,
Michel.

mpromonet, 2018-12-08:
Hi Denzel,

Cross-compiling WebRTC is not so trivial, I spent long time to build it especially for armv6, you can get the options from the circleci build https://github.com/mpromonet/webrtc-streamer/blob/master/.circleci/config.yml.

Best Regards,
Michel.

closed 2019-01-26 by mpromonet