mpromonet/webrtc-streamer #179: Link Error "undefined reference to Json::Value:***"

msedv, 2018-12-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mpromonet, 2018-12-10:
Hi Markus,

I see 2 classic possibilities either you didnot build ninja `jsoncpp` target or you build webrtc option `use_custom_libcxx=false` 

Best Regards,
Michel.

msedv, 2018-12-11:
Hi Michel,

thank you for your response, but none of my experiments in this direction worked. But let me ask from another ppoint of view: which Linux would you recommend me to install a complete build chain for webrtc-streamer running on a 64bit Linux - and is there eventually a more detailed instruction than the one in README.md?

Thank you!

Markus

mpromonet, 2018-12-11:
Hi Markus,

Then you should look like to symbols defined in the webrtc libraries, using nm and c++filt.
Using make `VERBOSE=1` could help also to get the link command.
The build on circleci use heroku:16 and appveyor use ubuntu 18.04.

Best Regards,
Michel.

closed 2019-01-26 by mpromonet