mpromonet/webrtc-streamer #180: Unable to build Ubuntu 18.04

journeym, 2018-12-19:
Followed the instruction form Readme.md
When i try to cmake . -DWEBRTCBUILD=Release -DWEBRTCROOT=/home/webrtc
I get 

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> WEBRTC_LIBRARY
>     linked by target "webrtc-streamer" in directory /home/rtsp/webrtc-streamer
> WEBRTC_PEERCONNECTION_LIBRARY
>     linked by target "webrtc-streamer" in directory /home/rtsp/webrtc-streamer
> WEBRTC_VIDEO_DECODER_LIBRARY
>     linked by target "webrtc-streamer" in directory /home/rtsp/webrtc-streamer
> WEBRTC_VIDEO_ENCODER_LIBRARY
>     linked by target "webrtc-streamer" in directory /home/rtsp/webrtc-streamer
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/rtsp/webrtc-streamer/CMakeFiles/CMakeOutput.log".
> See also "/home/rtsp/webrtc-streamer/CMakeFiles/CMakeError.log".

Im not very experienced with unix building, can you please point me in the right direction.

I also tried to follow the build procedures form CircleCI, and stuck at make all tgz
i get

BasicUsageEnvironment.hh : No such file or directory

abhijitnathwani, 2018-12-24:
Hi @journeym ,

Even I am facing this issue? @mpromonet can you help us out here? I tried building manually as well as through docker. Both the time facing same issues.

Abhijit

mpromonet, 2018-12-24:
Hi,

> WEBRTC_LIBRARY
linked by target "webrtc-streamer" in directory /home/rtsp/webrtc-streamer

means the library webrtc is not found in  `${WEBRTCROOT}/src/out/${WEBRTCBUILD}/obj`
Maybe you ran cmake, interrupt it during webrtc build and run another time cmake.

In such a case, you should remove `${WEBRTCROOT}/src/out/${WEBRTCBUILD}`

Best Regards,
Michel.

abhijitnathwani, 2018-12-26:
Hi @mpromonet ,

I did the same way you asked, but the problem still persists. 

Abhijit

journeym, 2018-12-27:
I did try to remove the referenced directories and start from scratch, with the same results.
Here is the contents of the directory you mentioned. Whats is a missing file name and location?
`root@ubuntu:/home/webrtc/src/out/Release/obj# ls -la
total 38560
drwx------ 23 root root     4096 Dec 19 15:39 .
drwx------  4 root root     4096 Dec 19 15:38 ..
drwx------ 13 root root     4096 Dec 19 15:39 api
drwx------  4 root root     4096 Dec 19 15:38 audio
drwx------  3 root root     4096 Dec 19 15:28 base
drwx------  4 root root     4096 Dec 19 15:29 build
drwx------ 12 root root     4096 Dec 19 15:39 call
drwx------  8 root root     4096 Dec 19 15:36 common_audio
drwx------  3 root root     4096 Dec 19 15:38 common_video
-rw-r--r--  1 root root 39253794 Dec 19 15:39 libwebrtc.a
drwx------ 13 root root     4096 Dec 19 15:38 logging
drwx------ 11 root root     4096 Dec 19 15:39 media
drwx------ 18 root root     4096 Dec 19 15:38 modules
drwx------  4 root root     4096 Dec 19 15:37 p2p
drwx------  4 root root     4096 Dec 19 15:39 pc
-rw-r--r--  1 root root        0 Dec 19 15:28 poison_audio_codecs.stamp
-rw-r--r--  1 root root        0 Dec 19 15:28 poison_software_video_codecs.stamp
drwx------ 28 root root     4096 Dec 19 15:37 rtc_base
drwx------  2 root root     4096 Dec 19 15:28 rtc_tools
drwxr-xr-x  2 root root     4096 Dec 19 15:35 sdk
drwx------  3 root root     4096 Dec 19 15:36 stats
drwx------  6 root root     4096 Dec 19 15:36 system_wrappers
drwx------  2 root root     4096 Dec 19 15:28 test
drwx------  5 root root     4096 Dec 19 15:28 testing
drwx------ 21 root root     4096 Dec 19 15:38 third_party
drwx------  5 root root     4096 Dec 19 15:39 video
-rw-r--r--  1 root root     3071 Dec 19 15:28 webrtc_common.ninja
-rw-r--r--  1 root root        0 Dec 19 15:36 webrtc_common.stamp
-rw-r--r--  1 root root   130814 Dec 19 15:28 webrtc.ninja
`

mpromonet, 2018-12-29:
Hi,
Cmake is looking for libwebrtc.a that seems to exists.
Maybe an acces problem, permissions ? 
Best regards
Michel

closed 2019-01-26 by mpromonet