mpromonet/webrtc-streamer #182: circleci build failure

dmathew93, 2018-12-21:
Hey Michel,

I decided to try and do the circleci build route and forked your repository. So far all 4 builds fail and in particular for the rpi_7 build it says 

`src/main.o: In function `main':
/webrtc-streamer/src/main.cpp:206: undefined reference to `cricket::StunServer::StunServer(rtc::AsyncUDPSocket*)'
collect2: error: ld returned 1 exit status
Makefile:97: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1
Exited with code 2`

none of the builds so far work for this in the circleci, any advice?

Denzel

dmathew93, 2018-12-21:
Unrelated note - been struggling to get this working but realized part of your circleci incorporated archiving the tgz: was able to grab the latest one (which idk why its failing on my builds) and transported it to my pi. It works!

I'm still curious to know the process on what im doing wrong on this but I'm now super happy that I can grab a working webrtc. Thank you so much Michel for your work on this.

mpromonet, 2018-12-22:
HI Denzel,

Using the latest WebRTC the StunServer is not build with `rtc_include_tests=false`, you could either turn `rtc_include_tests=true` or remove the StunServer creation.

Best Regards,
Michel.

commit 0bf18589 (2018-12-22): fix #182: update build with latest WebRTC
commit 80b2709d (2018-12-22): fix #182: update build with latest WebRTC
commit 8f0ee059 (2018-12-22): fix #182: update build with latest WebRTC
commit b344d83a (2018-12-24): fix #182: update build with latest WebRTC
closed 2018-12-22 by mpromonet